richonguzman/LoRa_APRS_Tracker #5: Settings for SH1106 display

Crille-ONCR16, 2023-06-26:
Dear, 
Would it be possible to get a file for using ttgo with a larger display SH1106?
Thank you very much!

richonguzman, 2023-07-10:
hi, did you resolve this?

I found that you can try to add this:

#define USE_SH1106

on the display.h or display.cpp to try

Crille-ONCR16, 2023-07-10:
Dear,

I'm on vacation now, wensday i try this out.

Thx

richonguzman, 2023-08-05:
good news!!!

SH1106 screen are now working also , go to "display.cpp" and uncomment line9 to make it work!

richonguzman, 2023-08-05:
as firmware 2023.08.05 this is available

closed 2023-08-05 by richonguzman

Crille-ONCR16, 2023-08-06:
Ok, i load the file and i try this out ..
Manny thanks!

ON3CRX

Crille-ONCR16, 2023-08-14:
Dear Ricardo,

Could you look at the settings, because the received signal is the distance
always indicated incorrectly on the igate ...
Example : tracker at a distance of 8km , is on the igate always around
5865km????
The trackers that are with github of OE5BPA do give the correct mileage ...

Manny thx
ON3CRX

richonguzman, 2023-08-14:
Maybe is it a station hearead before having a fix on the gps?

Message ID: ***@***.***>

Crille-ONCR16, 2023-08-15:
It happens with several trackers, ON4CDR has this problem also ...
Personally, I have this with my tracker too.
Both have a good gps signal +12 ...

richonguzman, 2023-08-15:
Does this error appears on the syslog info? Or where did you see this on the igate?

I did not get this before as this issue was for the tracker.

Message ID: ***@***.***>

Crille-ONCR16, 2023-08-15:
I don't have a syslog server yet , I see this on the igate display and also
on the tracker menu 2 Listen trackers ..
i send you pictures with ON4CDR who is on 5600km and ON3MR at 5 km ( he use
the github from OE5BPA )
ON4CDR see this to that when i transmit the km's is wrong ...