richonguzman/LoRa_APRS_Tracker #6: compatibility with OE5BPA code

jranma, 2023-07-16:
Hello, 
Is interoperability ensured with hardware that runs on OE5BPA code? 
For example, is it possible to receive, with your i-gate code, the positions of a tracker operating with the OE5BPA code? 
And also in the other direction (is the position sent by your tracker correctly received on an OE5BPA igate)? 
I think it's an important point to ensure success with lora APRS development. 
Thanks for the great work! 73s
Julien HB9HRD

richonguzman, 2023-07-16:
Hi Julien,

as far as I know , all LoRa APRS firmwares have interoperability as they share "lora preamble bytes" and also the APRS way of sending packets. 

So yes, this Tracker Firmware will be received also by OE5BPA firmware iGates. And my iGate Firmware will receive also the Trackers with OE5BPA firmware. 

73

Mane76, 2023-07-17:
Hi Julien,

both, the software of the tracker and of the iGate from Ricardo, is fully backward compatible to OE5BPA in all combinations.

'73
Mane76

jranma, 2023-07-17:
Great news ! 
I'm of the opinion that this compatibility should be retained as far as possible, and at least for the most basic functions, to avoid the risk of system segmentation. 

Thank you for these excellent developments. I'll try to promote the network in Switzerland. 
73 Julien

closed 2023-07-17 by jranma