richonguzman/LoRa_APRS_Tracker #8: Support also for SX1268 would be great :-)

DC4FRT, 2023-07-18:
When ordering the boards of the Lora T-Beam, I paid attention to the frequency 430MHz, but unfortunately did not see the difference between SX1278 and SX1268. 
Now I have some SX1268 boards here in my hand which unfortunately do not work. 
Failure is "Error - Starting LoRa failed! "

I have also SX1278, they work really fine :-) great Job from Ricardo.

thanks and best regards
73, Thomas

jranma, 2023-07-21:
@richonguzman Can you please confirm that SX1268 boards do not work ? 
Would be useful, cheers 73 Julien

richonguzman, 2023-07-21:
Sx1268 chips will work in the future , right now they don’t (I need one board to further test it )

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

jranma, 2023-07-21:
@richonguzman Thanks for your great work. I sponsored you as a modest contribution to buy the card ;) 
Best 73, Julien

richonguzman, 2023-07-21:
Great

Will search the way to get one and update all to work

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

richonguzman, 2023-08-05:
I received the board and currently working on it , wait just a little!

richonguzman, 2023-09-26:
hi, new version specially for SX1268 board is available now... please test as it is running smooth now for me

please test and confirm!

kratinoska, 2023-10-11:
I got board with SX1268 and seems to me that it does not work well. On radio listening 433.775 is quite different against ordinary T-beam with SX1278 and igate received one packet on power on only. see:
https://aprs.fi/#!mt=roadmap&z=11&call=a%2FOK2MNL-16&timerange=3600&tail=3600

Mane76, 2023-10-11:
@kratinoska , please consider two things
1) using aprs.fi for determining if an specific iGate has rx you is not feasible. If several iGates do rx you only the "fastest" one will be shown on aprs.fi
2) sx1268 is generally more sensitive in rx compared to sx1278, but does not tolerate big frequency offsets. sx1268 may tolerate +/-3kHz, sx1278 +/-10kHz or even more

For evaluation use Ricardos iGate connected to serial console and transmit with the tracker. Each frame should be visible in the serial console, including the offset.

kratinoska, 2023-10-12:
In Brno city we have few iGates based on OE FW.
My T-Beam with SX1268 and Ricardo FW:
I turn on T-beam and only one pocket (possition) was received and shown on Aprs.fi
https://aprs.fi/#!mt=roadmap&z=11&call=a%2FOK2MNL-16&timerange=3600&tail=3600
T-beam was on during my way to work (approx 20 minutes). And no other "mark" shown on maps....

richonguzman, 2023-10-12:
can you explain it better, I did not get your testing scenario or your configuration of the tracker

kratinoska, 2023-10-18:
probably bad koax connection between t-beam and antena, ...

kratinoska, 2023-10-18:
confirmed. With new cable it works very well.