richonguzman/LoRa_APRS_Tracker #9: Support for AXP2101

jranma, 2023-07-21:
Hello
Unfortunately, this code will not work on newer T-Beam boards sold by Lilygo (official Aliexpress store)

Since June 2023, AXP192 are no longer produced (or sold ?). New Lilygo T-beam boards are shipped with the new AXP2101.
The battery voltage stays at "0V" and the tracker turns on but has malfunctions.
As it is no longer possible to obtain modules with AXP192 from Lilygo, this problem will concern more and more new users.

Best 73, Julien HB9HRD

richonguzman, 2023-07-21:
Hi!

This was taken care on the env:TTGO ..... v1_2

This will make the new board with AXP2101 work ;)

Let me know how it goes

jranma, 2023-07-21:
works perfectly, thanks a lot

closed 2023-07-21 by jranma