richonguzman/LoRa_APRS_Tracker #11: Static IP address

tk5ep, 2023-07-22:
Hi Ricardo,
Not a issue, but an improvement request.

It would be nice to have the choice between DHCP IP address or static. ;-)

73 Patrick TK5EP

richonguzman, 2023-07-22:
Hi Patrick, why is this needed?

This gets in conflict somehow with the OTA part and also the ip could be defined easy in the WiFi router

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

tk5ep, 2023-07-23:
Hi Ricardo,

Why ?
It's sometimes needed in a  private network where there is no DHCP for security reason and you don't have other choice.

Doing it on the device side is much easier than on the router side, as you need to know the MAC Address of the module.  You also need to have a sysop access to the router, which is not always possible.

And when obliged to use static addressing, if there is a conflict with OTA, it's to the user to take care of this.

73 Patrick.

closed 2023-09-11 by richonguzman