richonguzman/LoRa_APRS_Tracker #12: parameters

jranma, 2023-07-25:
Hello, 
Most of the parameters are intuitive but could you describe precisely what these parameters mean: 

- nonSmartBeaconRate
- rememberStationTime
- maxDistanceToTracker
- standingUpdateTime

Would be grateful, thanks !

mpbraendli, 2023-07-26:
I was wondering about `maxDistanceToTracker`, so I had a look at the source code for other parameters you mention too.

- `nonSmartBeaconRate` is the transmit interval when `smart_beacon.active` is false.
- `rememberStationTime` is the duration for how long stations are remembered. This is used for showing the list in the LISTENING> display.
- `maxDistanceToTracker` is not used anywhere.
- `standingUpdateTime` is an interval that does something with how the altitude is encoded, if you didn't move since the previous report... But I'm not entirely sure. It's related to the APRS spec 1.0.1 page 48:

> In the special case of `c` = " " (space), there is no course, speed or range data, in which case the `csT` bytes are ignored.

But as said, I'm not entirely sure and Ricardo hopefully has more details or corrections.

richonguzman, 2023-07-27:
Hi,

* nonSmartBeaconRate: time for fixed time beacon when no SmartBeacon is active
 
* rememberStationTime: @mpbraendli was right , time for remembering a station/tracker in the listening part

* maxDistanceToTracker: this is an idea not implemented but somehow the tracker should only interact with trackers nearby, this is why this was an idea still

* standingUpdateTime: when 15 min has happened and the tracker hasn't move away from the center of the last point on gps for the minDistTx, it will send a beacon at 15 min. but sending the first "csT" byte (c) as " ". This makes APRS-IS servers interpret that the station has no movement and this shows not speed, course , etc