richonguzman/LoRa_APRS_Tracker #15: Number of satellites in comments or status

rippervii, 2023-07-30:
Hi, 

Just a proposal to add a function to show number of satellites received in commends or status.

Kind regards,

Stijn

richonguzman, 2023-07-30:
hi Stijn,

the main goal of all the APRS LORA project is to be able to make all the communication as clean and the lowest data as posible. So I hope you understand why I've not added a lot of info that could be shown on the comment section of the LoRa Beacon.

Some asked for RSSI, SNR, Satellites, comments super long and so on...

please tell me why this is so important besides just a few test for validation of your tracker (which I could help also to be satisfied with the info you get)

remember: I didn't put this as It would take to long to config every variable put on the comment .

rippervii, 2023-07-30:
This is for coverage tests. 
I thinks 3 characters,  for example s=7, is shorter and cleaner then a whole github url.

richonguzman, 2023-07-30:
> This is for coverage tests.

I could add this on yours specially, let me work on this in the afternoon

> I thinks 3 characters, for example s=7, is shorter and cleaner then a whole github url.
If you mean the status (this is sended only once as the tracker boots)  and never again

closed 2023-11-10 by richonguzman