richonguzman/LoRa_APRS_Tracker #16: Speed aprs.fi

rippervii, 2023-07-30:
Hi,

Speed is not always shown in aprs.fi
The speed only shows when standing still, not when moving.

Kind regards,

Stijn
On3rip

richonguzman, 2023-07-30:
Stijn 

can you please explain the "no speed" issue?

every gps beacon that the tracker sends has a position data (encoded) and also current speed and course/altitude also encoded.

please give further information

rippervii, 2023-07-30:
I'll try to get some screenshots. Give me some time ;)
It is the current speed thats not decodes by aprs.fi I think. 
The reason why I don't now. 
With other tracker programs this works fine.

richonguzman, 2023-07-30:
the tracker is configured to show Current Speed +Course or CurrentSpeed+Altitude

one of both is always sended.

take notice that if the tracker is not moving for 15 min the beacon packet has "zero" speed shown ... it is sending the speed info on the aprs beacon packet, but with a zero value or null value.

let me know which callsign to look to give you more info

rippervii, 2023-07-30:
We noticed this problem during last night with on4cdr callsign

richonguzman, 2023-07-30:
ON4CDR-8 is showing on each beacon point on APRS.Fi for the last 24 hrs a speed info correctly , maybe we are looking at different data?

richonguzman, 2023-08-08:
did it happen again?

richonguzman, 2023-08-26:
hello?

closed 2023-09-11 by richonguzman