richonguzman/LoRa_APRS_Tracker #19: Unable to compile tracker for TTGO T-Beam v0.7

btor, 2023-08-03:
Definitions for Power saving mode missing.

Compiling .pio\build\ttgo-t-beam-v0_7\src\station_utils.cpp.o
In file included from src/LoRa_APRS_Tracker.cpp:16:
src/power_utils.h:16:14: error: 'TwoWire' has not been declared
   bool begin(TwoWire &port);
              ^~~~~~~
In file included from src/power_utils.cpp:1:
src/power_utils.h:16:14: error: 'TwoWire' has not been declared
   bool begin(TwoWire &port);
              ^~~~~~~
src/power_utils.cpp:10:29: error: 'bool PowerManagement::begin' is not a static data member of 'class PowerManagement'
 bool PowerManagement::begin(TwoWire &port) {
                             ^~~~~~~
src/power_utils.cpp:10:29: error: 'TwoWire' was not declared in this scope
src/power_utils.cpp:10:38: error: 'port' was not declared in this scope
 bool PowerManagement::begin(TwoWire &port) {
                                      ^~~~
src/power_utils.cpp:10:38: note: suggested alternative: 'word'
 bool PowerManagement::begin(TwoWire &port) {
                                      ^~~~
                                      word
src/power_utils.cpp: In member function 'double PowerManagement::getBatteryVoltage()':
src/power_utils.cpp:138:1: error: no return statement in function returning non-void [-Werror=return-type]   
 }
 ^
src/power_utils.cpp: In member function 'double PowerManagement::getBatteryChargeDischargeCurrent()':        
src/power_utils.cpp:151:1: error: no return statement in function returning non-void [-Werror=return-type]   
 }
 ^
src/power_utils.cpp: In member function 'bool PowerManagement::isBatteryConnected()':
src/power_utils.cpp:160:1: error: no return statement in function returning non-void [-Werror=return-type]   
 }
 ^
src/power_utils.cpp: In member function 'bool PowerManagement::isChargeing()':
src/power_utils.cpp:169:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
cc1plus.exe: all warnings being treated as errors
In file included from src/menu_utils.cpp:7:
src/power_utils.h:16:14: error: 'TwoWire' has not been declared
   bool begin(TwoWire &port);
              ^~~~~~~
In file included from src/station_utils.cpp:5:
src/power_utils.h:16:14: error: 'TwoWire' has not been declared
   bool begin(TwoWire &port);
              ^~~~~~~
Compiling .pio\build\ttgo-t-beam-v0_7\libed7\TinyGPSPlus\TinyGPS++.cpp.o
Compiling .pio\build\ttgo-t-beam-v0_7\lib54d\Time\DateStrings.cpp.o
Compiling .pio\build\ttgo-t-beam-v0_7\lib54d\Time\Time.cpp.o
Compiling .pio\build\ttgo-t-beam-v0_7\lib3f2\OneButton\OneButton.cpp.o
*** [.pio\build\ttgo-t-beam-v0_7\src\LoRa_APRS_Tracker.cpp.o] Error 1
*** [.pio\build\ttgo-t-beam-v0_7\src\power_utils.cpp.o] Error 1
*** [.pio\build\ttgo-t-beam-v0_7\src\menu_utils.cpp.o] Error 1
*** [.pio\build\ttgo-t-beam-v0_7\src\station_utils.cpp.o] Error 1
======================================== [FAILED] Took 47.59 seconds ========================================

Environment       Status    Duration
----------------  --------  ------------
ttgo-t-beam-v0_7  FAILED    00:00:47.590
=================================== 1 failed, 0 succeeded in 00:00:47.590

richonguzman, 2023-08-03:
Hi,

are you choosing specially V0_7 on VSCODE and importing all through Platformio?

btor, 2023-08-04:
Hi,
Thanks for the fast response.
Yes, I use VSCODE with Platformio and V0_7 is selected before compilation.
Power saving is not implemented in power_utils.cpp for V0_7
Best Regards,
Bela

btor, 2023-08-07:
Hello Ricardo,

I added necessary definitions to power_utils.h and power_utils.cpp required for a succesful compilation.
Since I have no information on V0_7 hardware, no power manadgement related code was added.
Included you'll find the zipped files.

Best regards,
Bela

[power_utils.zip](https://github.com/richonguzman/LoRa_APRS_Tracker/files/12284880/power_utils.zip)

richonguzman, 2023-08-08:
this is added now to the next firmware available and compiles good, so it should work now, thanks a lot

richonguzman, 2023-08-08:
should the V0_7 be called on the powermanagment::setup() also?

btor, 2023-08-10:
No there is nothing to initialize in setup()
I added battery voltage measurement to power_utils.cpp and menu_utils.cpp and comments were added regarding v0.7 to power_utils.cpp, power_utils.h
Included you'll find the mdified files.
[src.zip](https://github.com/richonguzman/LoRa_APRS_Tracker/files/12312419/src.zip)

richonguzman, 2023-08-10:
could you confirm it works for you with this mods?

btor, 2023-08-10:
Yes, it works, I tested the mods with a V0.7 board.
Best regards,
Bela / HB9HEY

richonguzman, 2023-08-10:
I uploaded a new firmware , could you please confirm it still runs well on yours?

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

richonguzman, 2023-08-11:
> I can't compile the new firmware for v0_7. I get the following compiler error:
>
> Compiling .pio\build\ttgo-t-beam-v0_7\src\lora_utils.cpp.o
> src/lora_utils.cpp:1:10: fatal error: RadioLib.h: No such file or directory
>
>
>   *   Looking for RadioLib.h dependency? Check our library registry!
>   *
>   *   CLI > platformio lib search "header:RadioLib.h"
>   *   Web > https://registry.platformio.org/search?q=header:RadioLib.h
>   *
>
>
> #include <RadioLib.h>
> ^~~~~~~~~~~~
> compilation terminated.
> Archiving .pio\build\ttgo-t-beam-v0_7\lib04d\libWire.a
> Compiling .pio\build\ttgo-t-beam-v0_7\libaa6\SPI\SPI.cpp.o
> Compiling .pio\build\ttgo-t-beam-v0_7\libbd1\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
> Compiling .pio\build\ttgo-t-beam-v0_7\libbd1\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
> Compiling .pio\build\ttgo-t-beam-v0_7\libbd1\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
> Compiling .pio\build\ttgo-t-beam-v0_7\lib078\Adafruit GFX Library\Adafruit_GFX.cpp.o
> Compiling .pio\build\ttgo-t-beam-v0_7\lib078\Adafruit GFX Library\Adafruit_GrayOLED.cpp.o
> *** [.pio\build\ttgo-t-beam-v0_7\src\lora_utils.cpp.o] Error 1
> ===================================== [FAILED] Took 19.76 seconds =====================================

Radiolib is only for sx1268 board .. so it should not be a problem with yours (v0_7)

Try deleting it at your PC on the lora_utils.cpp and platformio ini to see how it goes

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

ttgo-t-beam-v0_7 FAILED 00:00:19.759
================================ 1 failed, 0 succeeded in 00:00:19.759 ================================

btor, 2023-08-11:
After adding defined(TTGO_T_Beam_V0_7)  in multiple paces in lora_utils.cpp and making "full clean" in Visual Studio Code
I could compile the current firmware for V0.7

The modifications I submitted yesterday -> display the Battery voltage on the display were not merged to this firmware. Should I implement it?

Best Regards,
Bela / HB9HEY

[lora_utils.zip](https://github.com/richonguzman/LoRa_APRS_Tracker/files/12321934/lora_utils.zip)

richonguzman, 2023-08-11:
I will do both today(night) and message you to test ;)

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

richonguzman, 2023-08-12:
it seems all was added correctly, please test the "new" Main 2023.08.12 firmware to validate it and share your callsign to add this to readme as to share your help ;)

btor, 2023-08-13:
I could compile the firmware for V0_7. Since I am currently abroad, I could not test the functionality with an igate. I'll test it next week when I am back home again.

I noticed a missing word in power_utils.h
the following section in the top of the file:
#ifdef TTGO_T_Beam_V0_7
// The V0.7 boards have power managment components connected to TwoWire. 

Please add a "no" between "have" and "power" like this:

#ifdef TTGO_T_Beam_V0_7
// The V0.7 boards have no power management components connected to TwoWire. 
//

The button does not work properly with my V0_7 board, it is almost impossible to navigate in and between the menus. Perhaps the button does not work properly, unfortunately, I can't test it at the moment. I'll test it wen I'm back home again.

Best Regards,
Bela HB9HEY

btor, 2023-08-14:
I solved the button issue with V0_7 board.. 
BUTTON_PIN was defined as 38 in pins_config.h, but the button is connected to pin 39 on V0_7 board

pins should be defined for the V0_7 board as follows:

#ifdef TTGO_T_Beam_V0_7
#define GPS_RX          15
#define GPS_TX          12
#define BUTTON_PIN      39
#endif

After changing the button definition the menu works as expected on my V0_7 board.

Kind regard,
Bela HB9HEY

richonguzman, 2023-08-14:
added the corrections

richonguzman, 2023-08-16:
Great, so it works and even the info on screen and the middle button works

Please confirm to close this issue

btor, 2023-08-17:
You can close this issue.
Thanks for the excellent support.
Kind regards,
Bela

closed 2023-08-17 by btor