richonguzman/LoRa_APRS_Tracker #21: Support for BME sensor

iw2ejh, 2023-08-11:
Do we have BME sensor support on the tracker as we have in the iGate? Eventually could we have it?
thanks
iw2ejh

richonguzman, 2023-08-12:
hi, yes we could, give me a few days, if it hasn't been done, let me know as this is good idea. but I'm finishing right now a few  ideas before

richonguzman, 2023-08-26:
please test now the firmware 2023.08.26 as it hast BME support for showing data on screen.... with this validated I will look for a way to send data also to APRS to have telemetry

ON2ON-HAM, 2023-08-27:
Testing 02.26 at this moment, works great, BME data is on the oled, alternating nicely with other data. 
So no comments/remarks at all !
Waiting for the Aprs/wx updater ;)

Take care
ON2ON-7

richonguzman, 2023-08-28:
Waiting for the Aprs/wx updater ??? please explain

you need the data sanded as telemetry from tracker?

ON2ON-HAM, 2023-08-28:
Hi,
It would be nice to have the WX data send to Aprs, like in the Igate. As I understand it , now the WX data is only on the local Oled of the tracker.  So the Tracker can behave like a CWOP node and tracker at the same time.

73 ON2ON

iw2ejh, 2023-08-29:
[image]
The reading is correct and appear in the display
I suspect that when the display is configured run in ecomode and shutdown, the BME is not working (I do not know how the ecomode is triggered, but sensor and display share the same pins)
Also would it be possible to add an offset for the pressure value so we can correct it based on the altitude of the wx station?
Thanks

iw2ejh, 2023-08-29:
Correction to my previous post, the value from BME is working even with the display set in ecomode, probably anothe glitch caused this behavior during tests. I flashed again the filesystem with various settings and now the values for temp, humidity and pressure apear correctly on the display even if it shutdown and up again.