richonguzman/LoRa_APRS_Tracker #23: Issue whith Lilygo T-Beam v1.2 with AXP2101

HJ5JMS, 2023-08-22:
Hello !

I recently received a Lilygo T-Beam v1.2 with AXP2101 from Aliexpres, but after installing the tracking software, the device doesn't work and seems to be related to the APX210.

The main screen shows battery: -0.00v and the device got stuck when preparing to transmit. Checking the serial monitor, I also found that there is an error message during boot: [ERROR] [AXP192] Boot failed!

Is there a way to handle this problem, or is the hardware definitely not supported yet?

[image]
[image]

Thank you!!

richonguzman, 2023-08-22:
hi , as you have noticed you need to use the "env:ttgo-t-beam-v1_2" 

this will enable the use of the AXP2101 power management chip and your Tracker should work fine now?

HJ5JMS, 2023-08-22:
Hello !

You're right, I was overlooking some important env - related steps in the flashing process.

Thank you!

closed 2023-08-22 by HJ5JMS