richonguzman/LoRa_APRS_Tracker #25: GPS send null position

iw2ejh, 2023-08-25:
I was in version 2023-7-13 and everything in the tracker was working as expected
I updated to version 2023-8-24
with that version, (after the GPS acquire the position, led blinking) the tracker send out a packet with null position.2023-08-25 12:59:39 CEST: [IW2EJH-5](https://aprs.fi/?c=raw&limit=&call=IW2EJH-5)>APLRT1,WIDE1-1,qAR,[iw2ejh-11](https://aprs.fi/?c=raw&limit=&call=iw2ejh-11):>https://github.com/richonguzman/LoRa_APRS_Tracker 2023.08.19
   type: status
   srccallsign: IW2EJH-5
   dstcallsign: APLRT1
   status: https://github.com/richonguzman/LoRa_APRS_Tracker 2023.08.19

2023-08-25 13:02:35 CEST: [IW2EJH-5](https://aprs.fi/?c=raw&limit=&call=IW2EJH-5)>APLRT1,WIDE1-1,qAR,[iw2ejh-11](https://aprs.fi/?c=raw&limit=&call=iw2ejh-11):!null7?TWQ$XR[E8Q Bat=4.20V (87mA) [Invalid packet]
   type: location
   srccallsign: IW2EJH-5
   dstcallsign: APLRT1
   messaging: 0

while with the previous FW, the packet looks like:
2023-08-25 12:46:13 CEST: [IW2EJH-5](https://aprs.fi/?c=raw&limit=&call=IW2EJH-5)>APLRT1,WIDE1-1,qAR,[iw2ejh-11](https://aprs.fi/?c=raw&limit=&call=iw2ejh-11):!/7?UVQ$Xo[EIQ Bat=4.12V (-90mA)
   type: location
   format: compressed
   srccallsign: IW2EJH-5
   dstcallsign: APLRT1
   latitude: 45.813525461638 °
   longitude: 10.070165858986 °
   altitude: 229.81562896278 m
   symboltable: /
   symbolcode: [
   messaging: 0
   posresolution: 0.291 m
   comment: Bat=4.12V (-90mA)

I rolled back to that version and it work fine, I put again the new version and got the same behaviour.

richonguzman, 2023-08-25:
Hi,

can you tell me which board are you using? 

my last firmware available for the Tracker is 2023.08.19 so this is strange

iw2ejh, 2023-08-25:
ops sorry, yes version is 2013-8-19,  i named the directory with the date I download the fw.  
The board is 'ttgo-t-beam-v1'  and fw is 2023-08-19

richonguzman, 2023-08-25:
could it be you flashed another filesystem? or changed something when uploading the new firmware?

which version was the "before" version which has no flaws for you?

iw2ejh, 2023-08-25:
version 2023-7-13  was the version I then updated to 2023-8-19 and the formed packet did not show the gps position.  I then rolled back to version 2023-07-13 and the gps position in the packet was correct.
I then updated again to 2023-08-19 and got again the null? in the packet sent 
you can see the trace in aprs.fi:  https://aprs.fi/?c=raw&call=IW2EJH-5   

Only thing i can have touched is the config file, i cup/paste the first part, I will try to reload with the original  and run a new test.

richonguzman, 2023-08-25:
Please update also the filesystem with the new data (not pasting the old) as new features has been added and this may be causing the problem

iw2ejh, 2023-08-25:
[image]

richonguzman, 2023-08-25:
Maybe wait a little longer and tell me about it later.

Usually the time is the first to be received , then the date and finally the gps fix (and position)

richonguzman, 2023-08-26:
did it got resolved?

iw2ejh, 2023-08-28:
The "null?" packet sent out to aprs in the packet string is solved.  The date  displayed on the Oled is still at 1999-12-31. This is not important at the moment, I will program a new board for a friend tomorow and check if it work. I will update when test done.

closed 2023-09-11 by richonguzman