richonguzman/LoRa_APRS_Tracker #26: T-Lora V2 1 1.6

sysadminpower2019, 2023-08-30:
Buenas,

este proyecto tiene soporte para el modulo T-Lora?
Lo compre en este link https://www.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.5.303d1802nFPDmv

Este modulo ya tiene soporte de meshtastic por si acaso.

Tambien la pregunta sobre la instalacion y configuracion del proyecto. Funciona igual que el proyecto original de Lora_Aprs, usando el Terminal directamente para compila el programa?

richonguzman, 2023-08-30:
hola, el modulo T-lora32 v2.1.6 que tu tienes sirve para hacer de Tracker solo si se usa como TNC hacia el teléfono celular.

sin GPS del telefono no tiene sentido que sea tracker si la placa no tiene GPS-
.

para la instalación y configuration por ahora se hace todo dentro de VSCODE

sysadminpower2019, 2023-09-07:
segun entiendo PlatformIO es una version open source and VScode entonces deberia funcionar.

La otra pregunta es por que no se le da soporte completo al modulo. Siempre se puede facilmente agregar un modulo gps adicional. Tambien aqui la pregunta si el punto del APRS es solo mandar informacion de posicion tipo GPS. Quizas eso fue algo interesante al principio del projecto APRS pero hoy dia es mas util usarlo como plataforma de transmision de datos digitales.

richonguzman, 2023-09-07:
hola no entiendo tu reclamo de no tener soporte completo y para no caer en simplificar o que no me entiendas bien o se entienda que no quiero me explicare:

el modulo que tu tienes ahi se usa regularmente de iGate porque no tiene GPS incluido. Mi respuesta anterior iba a que si lo usas como modulo LoRa TNC al celular, desde Android+APRSDroid puedes enviar el punto de gps desde el teléfono hacia el modulo lora y así enviar tu posición.

por otro lado, si puedes usar el modulo de iGate como tracker pero tendrías que si o si agregarle el modulo de GPS aparte y habría que agregar codigo para eso.

te refieres a lo ultimo que es lo que deseas?

closed 2024-01-05 by richonguzman

sysadminpower2019, 2024-02-22:
si quieres manda me tu nombre en discord y conversamos a ver como te puedo ayudar con la documentacion @richonguzman