richonguzman/LoRa_APRS_Tracker #28: Button OFF  Function ?

DL4SDR, 2023-09-04:
If possible  in one of the next updates: Please an Option to Disable  BUTTON Function  due to use it only in one Mode

richonguzman, 2023-09-04:
Hello

Do you mean disable all buttons or just disable middle button?

DL4SDR, 2023-09-04:
Hi Ricardo, Middle Button.Aktuell testing dk4sdr-7...See aprs.fi73Herbert

DL4SDR, 2023-09-04:
> Ok
>
> But now you should explain me what should the middle button do : only TX? Or change callsigns? Or just to block menu options?

Hello Ricardo

 

>>> Or change callsigns

 

a friend of mine has some difficulties to use his Finger's (diabetic polyneuropathia) AND  he take the Tracker in his Jacket  Or  basket

 

hope it is makeable

 

73's

Herbert

richonguzman, 2023-09-04:
So he only need to disable middle button completely to avoid pressing it on the bag , right?

DL4SDR, 2023-09-04:
YES :-)

richonguzman, 2023-09-04:
Please mail me to show you what to change in the code for this custom version ;)

closed 2023-09-11 by richonguzman