richonguzman/LoRa_APRS_Tracker #32: temp not transmitted

airboxed, 2023-09-21:
At first many thanks for the nice project,
BME280 works, shows temp. pressure. ttc on screen.
but not transmitting to aprs.
regards dan 
vy 72 de DL6OW

richonguzman, 2023-09-21:
Hi, so far the wx data is not Tx on purpose.  If we send it we would need to do a few changes, wx telemetry/data is only processed when:
- send no more than 1 packet each 9-10 min
- change symbol 

If you are ok with it , it could be added ;)

airboxed, 2023-09-21:
Thanks for megafast reply. I am OK ! Regards Dan.

richonguzman, 2023-09-21:
"I'm ok" = "not now" or "please add this" ? 😅

airboxed, 2023-09-24:
Pleeeeeeaaaaase add ! Many thanks in advance! Dan

airboxed, 2023-09-27:
Pleaaaaasseee add. Many thanks in advance.
Dan

richonguzman, 2023-10-01:
Lets go DAN, time to test it , its ready and added!

closed 2023-10-01 by richonguzman

airboxed, 2023-10-03:
All things work very well.
Thanks and 72 de Dan!