richonguzman/LoRa_APRS_Tracker #33: On display the date is not updating.

iw2ejh, 2023-09-27:
In the display, the date is not updating. it is fixed at 1999-12-31 . the time is correct, the Lat Long is correct and the Locator is correct. Only date is wrong. (verified on two different trackers)

richonguzman, 2023-10-01:
did you test this more? as this is really strange behavior

OK1FWG, 2023-10-02:
Did you had to reset the GPS? If so, it can happen, that the GPS is not sending NMEA message containing the date,

iw2ejh, 2023-10-03:
Yes, on both devices, the GPS has been reset with the available tool. Do you know how to recover from this situation?

OK1FWG, 2023-10-03:
You've probably used something like the TTGO_T-beam_GPS-reset utility.
I don't have my code at hand right now, but as I remember, the reset utility enables only the NMEA_GGA sentence, which doesn't include the date.
`myGPS.enableNMEAMessage(UBX_NMEA_GGA, COM_PORT_UART1);`

I had to add to it something like this:
`myGPS.enableNMEAMessage(UBX_NMEA_RMC, COM_PORT_UART1);`

Then I've got the time displaying correctly. I'll look at my code at home, please QRX.

iw2ejh, 2023-10-03:
Yes Willi,  you got the point, I added the RMC sentence after the gps reset and bingo, this unlocked the situation, now the Date appear correctly on the display.

iw2ejh, 2023-10-03:
Problem description:  Only Date is not updated on the display
Cause: have used the GPS reset tool to force GPS to send again NMEA sentences after Meshstastic was loaded on the TTGO board (it use different protocol)
Solution:  modification of the reset tool to include RMC sentence in addition to the GGA one
------------------------------------------------------------
 do {
    if (myGPS.begin(SerialGPS)) {
      Serial.println("Connected to GPS");
      myGPS.setUART1Output(COM_TYPE_NMEA); //Set the UART port to output NMEA only
      myGPS.saveConfiguration(); //Save the current settings to flash and BBR
      Serial.println("GPS serial connected, output set to NMEA");
      myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSA, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSV, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_VTG, COM_PORT_UART1);
      myGPS.enableNMEAMessage(UBX_NMEA_RMC, COM_PORT_UART1);  <------
      myGPS.enableNMEAMessage(UBX_NMEA_GGA, COM_PORT_UART1);
      myGPS.saveConfiguration(); //Save the current settings to flash and BBR
      Serial.println("Enabled/disabled NMEA sentences");
      break;
-------------------------------------------------------------

OK1FWG, 2023-10-03:
> Yes Willi, you got the point, I added the RMC sentence after the gps reset and bingo, this unlocked the situation, now the Date appear correctly on the display.

I am glad I could help. Enjoy it!

richonguzman, 2023-10-03:
thanks a lot for this!

will try to save the info on the WIKI also!

richonguzman, 2024-03-31:
i had this all missing!! until I needed to reset one T-beam

so i will upload this all new into a new updated repository

closed 2024-03-31 by richonguzman