richonguzman/LoRa_APRS_Tracker #34: Battery voltage/current not working

lostdog2, 2023-10-09:
I uploaded in the LILYGO T_BEAM433 MHz.
The version of the board is a mystery:
- the white label say '2ASYE-T-BEAM MODEL V1.1'
- the text on PCB close to the antenna say 'AXP2101-V1.2'
On the display, the voltage is always 0, current is always -3 mA.
I tried to upload some different version of firmware, but the battery data are not showed.
For some other firmwares I used the 1.2 version and is working.

Somebody can help me?

Many thanks

richonguzman, 2023-10-09:
hi!

https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/3.-Supported-Boards-and-Environment-Selection#where-do-you-select-the-environment

there you can select the board you have (V1_2) and upload it to work good.

lostdog2, 2023-10-10:
richonguzman,

Many thanks for you reply.
I'm a newbie with vscode, platformio and T-BEAM.

So, I selected the firmware version but not the correct environment...

Now the battery voltage and percentage are correctly displayed.

Have a good day

Lostdog

closed 2023-10-10 by richonguzman