richonguzman/LoRa_APRS_Tracker #39: smart_beacon

kratinoska, 2023-10-17:
I have set active to false in first beacon and true in second, but when I switch T-beam to second beacon, monitor on USB says [INFO][Main] Smart Beacon is: Off, but should be ON.

"beacons": [
		{
			"callsign": "OK2MNL-14",
			"symbol": "[",
			"overlay": "/",
			"comment": "",
			"smart_beacon": {
				"active": false,
				"slowRate": 120,
				"slowSpeed": 3,
				"fastRate": 60,
				"fastSpeed": 15,
				"minTxDist": 50,
				"minDeltaBeacon": 20,
				"turnMinDeg": 12,
				"turnSlope": 60
			}
		},
		{
			"callsign": "OK2MNL-14",
			"symbol": ">",
			"overlay": "/",
			"comment": "",
			"smart_beacon": {
				"active": true,
				"slowRate": 120,
				"slowSpeed": 10,
				"fastRate": 60,
				"fastSpeed": 70,
				"minTxDist": 100,
				"minDeltaBeacon": 12,
				"turnMinDeg": 10,
				"turnSlope": 80
			}
		},

richonguzman, 2023-10-17:
Hi, 

do you mean that when you change between calligns (saved on the tracker) it won't change their smarbeacon state? or that the message shown in serial output is faulty?

kratinoska, 2023-10-18:
it won't change their smarbeacon state

paquio, 2023-12-04:
Ciao, HI, IZ0JHY Antony 

ho caricato lo skect, tutto ok ma lo smartbeacon non funziona l'icona sulla cartina aprs rimane statica e non gira secondo la direzione i dati gps arrivano legorarmente il led rosso lapeggia +8 sattelliti in vista mi puoi aiutare? grazie

paquio, 2023-12-04:
> Hi
>
> Are you moving when you say beacons are not being transmitted?
>
> Did you try with just fixed beacon time?

Come posso risolvere 

⁣Ottieni BlueMail per Android ​

richonguzman, 2023-12-04:
Find me on telegram as @richonguzman to keep on resolving this ;)

closed 2023-12-08 by richonguzman