richonguzman/LoRa_APRS_Tracker #43: Enable GPS RTC-battery charging for AXP2101 board

longview, 2023-11-27:
Hi, love the project!

I noticed the RTC battery for the GPS module on the V1.2 T-Beam board is not charged by default.

For the AXP2101 boards this can be enabled by calling:
`PMU.setButtonBatteryChargeVoltage(3300);` 
followed by:
`PMU.enableButtonBatteryCharge();`
Which I suggest enabling in PowerManagement::activateGPS()

For the older boards I think enabling LDO1 to around 3.3 V would work the same.
The battery (https://www.sii.co.jp/en/me/datasheets/ms-rechargeable/ms412fe-5/) can be charged to somewhere between 2.8-3.3 V, and a silicon diode in series will drop the voltage by around 0.3 V or so.

There are some functions to read the voltage as well, so in principle the software could enable charging only when needed but I think the power draw is so small that it shouldn't matter.

richonguzman, 2023-11-28:
hi, could you test it ?

the goal of this is to have a faster GPS fix?

longview, 2023-11-28:
Yes, I've already tested the change and it immediately charged the battery (it only takes a few minutes to reach full charge).

The difference with a charged battery is that the time to first fix _may_ be faster, especially if the tracker is quickly powered on/off. 
For a quick power cycle (e.g. when uploading new code) the fix-time is very short now.
Today I left it off for around 8 hours and the clock was correct, but it still needed a few minutes to get a fix just like a cold start.

So it's not a massive improvement over all but I think it's worth keeping the battery charged since at least the clock will be correct on startup.

richonguzman, 2023-11-29:
hi, 

what I have seen is that the gps antenas of v1.2 are not that good as the old in v.1.0

on my V1.2 I have the v.1.0 antena of gps and without any of this mods the gps is fix on less than a minute and showing easily one or two more satelites than new gps antena

closed 2023-12-13 by richonguzman