richonguzman/LoRa_APRS_Tracker #44: BMP280 MOD

DG1FI, 2023-11-29:
File: bme_utils.cpp

Here is my CODE-MOD to disable transmit huminidy data for BMP280

`String readDataSensor(String type) {
    String wx, tempStr, humStr, presStr;
    float newTemp   = bme.readTemperature();
	float newHum;
	
	#ifndef BMPSensor
		newHum    = bme.readHumidity();
	#else
		newHum = 0;
	#endif 
    
	float newPress  = (bme.readPressure() / 100.0F);
    
    if (isnan(newTemp) || isnan(newHum) || isnan(newPress)) {
      Serial.println("BME280 Module data failed");
      if (type == "OLED") {
        wx = " - C    - %    - hPa";
      } else {
        wx = ".../...g...t...r...p...P...h..b.....";
      }
      return wx;
    } else {
      tempStr = generateTempString(newTemp, type);
    
		#ifndef BMPSensor
			humStr  = generateHumString(newHum, type);
		#else
			humStr  = "-99";
		#endif
      
	  presStr = generatePresString(newPress + (Config.bme.heightCorrection/CORRECTION_FACTOR), type);
      if (type == "OLED") {
        #ifndef BMPSensor
			wx = tempStr + "C   " + humStr + "%   " + presStr + "hPa";
		#else
			wx = "T: " + tempStr + "C " + "P: " + presStr + "hPa";
		#endif
      } else {
        #ifndef BMPSensor
			wx = ".../...g...t" + tempStr + "r...p...P...h" + humStr + "b" + presStr;
		#else
			wx = ".../...g...t" + tempStr + "r...p...P...h..b" + presStr;
		#endif
	  }
      return wx;
    }
  }`

radiohound, 2023-12-09:
Thanks for this. Where and how do we define BMPSensor?

richonguzman, 2023-12-12:
please test "bmp add-on" branch to validate the mods

radiohound, 2023-12-12:
Thank you! The temperature and pressure are showing correctly on the display. But it is not getting sent out as telemetry. I have set sendTelemetry to true.

richonguzman, 2023-12-12:
What does raw packet show on aprs.fi?

radiohound, 2023-12-12:
it shows:2023-12-12 05:49:25 PST: [K6ATV-13](https://aprs.fi/?c=raw&limit=&call=K6ATV-13)>APLRT1,WIDE1-1,qAR,[K6ATV-10](https://aprs.fi/?c=raw&limit=&call=K6ATV-10):!/;hr</fWP_!!Q

richonguzman, 2023-12-12:
hi only GPS data is showing for this station.

did you enable sending Wx data over RF in the tracker_conf.json ??

it sends once 10min a wx beacon packet, the other packets are always gps only(as APRS.fi processes it in no less than 10min)

radiohound, 2023-12-12:
Sorry, I just was not patient enough. After 10 minutes, the WX data was sent. Thank you so much for the help!

closed 2023-12-12 by richonguzman