richonguzman/LoRa_APRS_Tracker #45: Issues when Installing on T-BEAM v1.2

sysadminpower2019, 2023-12-07:
Hola,

por fin puede comprar un modulo nuevo con GPS y demas.

Trate tu version y le version original del software y no funciona.

Cuando instalo tu version en la pantalla me sale solo LORA APRS
(TRACKER)
etc.

Pero de ahi no funciona nada mas.

Si cambie el callsign. Ademas para estar seguro modifique el platform.ini para que solo este la version 1.2

El model exacto dice 2ASYE-T-Beam
T-Beam-AXP2101-v1.2

sysadminpower2019, 2023-12-10:
ya ahora funciono, tuve que usar VSCODE directo en linux. 
Creo que lo que faltaba es es parte del filesystem. 
Tengo que investigar como se puede hacer esa parte directamente desde la linea de comandos

richonguzman, 2023-12-11:
si averiguas me ayudarías tambien ya que seria genial poder hacer algunos script directos sin pasar por programas adicionales

sysadminpower2019, 2023-12-14:
lamentablemente no funciona al final, la placa nunca se conecta al GPS y los botones no sirve. Basicamente solo muestra la pantalla principal con el indicativo y nada mas

richonguzman, 2023-12-14:
Usaste el ENV que corresponde a la placa?

Subiste el firmware y filesystem después de seleccionar el ENV correcto?

closed 2023-12-19 by richonguzman

sysadminpower2019, 2024-01-04:
Ja escribieron en otro post aca en github que es un problema con esta
version de la placa por que hubo algun cambio. Asi que seria cuestion de
revisar el codigo fuente

https://ci6.googleusercontent.com/proxy/aB9cP4P8LChOK-Z8V_xjFErpus-Tz6mtUcod7ZuaXG7EIVIrDkWUUzd8Plo6j7_eLSedv3oHjlPor1j_cX6GdPP9F3DS_yAh_qMcO8taKQKVE4JJWpm8ufm0M1ZzBU-qbYMpqy5e8hbNOXY7cr0UEQ9AQ59-Uojy_6GFC2m-bE5rOu_RKycaO576pKwrhMRJJDLpslYk89Ff7dmm-RsVrihmNCqi9NK1CECTOfYS1uef0lNfFFCyOciiOW0UPlcDTRwm4aGPvnl6TL-h=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABXOLE643A52K7VQIJ3YKIA45A5CNFSM6AAAAABALZLPEGWGG33NNVSW45C7OR4XAZNWJFZXG5LFIV3GK3TUJZXXI2LGNFRWC5DJN5XKUY3PNVWWK3TUL5UWJTYAAAAAFIJ2YE4Q.gif>Message
ID: <richonguzman/LoRa_APRS_Tracker/issue/45/issue_event/
***@***.***>

richonguzman, 2024-01-04:
en el Wiki puse como se elige el firmware especifico con las características para cada placa, asi que siempre es bueno revisar ahi para cualquier consulta ;)

sysadminpower2019, 2024-01-04:
No es eso.
Aqui te copio el mensaje

I have the same problem with the new Chipset.
Seems the control of the chipset differs to the old version.
It hangs up during TX
AXP2101 - Power Managment Chipset
Platform: TTGO TBEAM-AXP2101 V1.2 Production Date labeled next to SMA
Connector 20230508.
Visual Studio Code Version : 1.85.1 mit Platform IO

Can i support in some way? Do you need further information?

> en el Wiki puse como se elige el firmware especifico con las
características para cada placa, asi que siempre es bueno revisar ahi para
cualquier consulta ;)
https://ci4.googleusercontent.com/proxy/ogRh33Gr_yNAkiGkkzZlkbuWbw5jZGcW_awxVkZ3QUr-bCZA2paMCUv9_MJh1pPd3U8o5Xj1m0jCJnEj1QkamVb2ArV8cjjVLKP2gIMUzkUYx_Q7MtTIMBnI_Ma_DMBtV3ClFBFLR_MqRrxWO0riYYTh6xY_jWTcJZ6ZqyQGFb6a6E_T2ICCF-uKJGCUSo6-q_WuhiWIEszWacHcUk5nyoD5Lwz0auNBjqlZLYlk4TpnL_cvOA=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABU3A5LWLJ22B2ZXXYDYM4ONPA5CNFSM6AAAAABALZLPEGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTP5TQ7I.gif>Message
ID: ***@***.***>

richonguzman, 2024-01-05:
Por eso te lo comento: 

en el WIKI sale como seleccionar la placa que usas (DEFAULT es T-Beam V1.2 que tiene AXP2101)
la antigua(T-Beam V.1) usaba AXP192

asi cuando seleccionas el ENVIRONMENT (que sale explicado como en la WIKI) le dices al programa/codigo de VSCODE que sepa que debe subir y que no de todo el firmware.

asi logramos tener muchas placas distintas corriendo el mismo firmware pero con las modificaciones puntuales para cada una. Como por ejemplo que unas funcionan con el chip de manejo de energia nuevo y otras el antiguo y otras con ninguno.

sysadminpower2019, 2024-01-19:
volvio a tratar y nada. El firmware si parece que carga bien y los menus estan todos ahi pero no se conecta al gps y no se como puedo probar si por lo menos se esta conectando con APRS. 
Me supongo que toda la informacion de fecha y hora se configuran por gps

sysadminpower2019, 2024-01-19:
al final parece que si conecta pero no completamente, si tiene la hora y fecha correcta pero el gps me muestra 0.00000 0.0000 for ejemplo

richonguzman, 2024-01-19:
Entonces si corre , solo le falta tiempo o nueva antena de GPS 

Cuando hay hora y fecha valida que si funciona , ahora es cosa de tiempo y tener un lugar despejado al cielo para que logre el fix de GPS

sysadminpower2019, 2024-01-22:
a la final si funciono. Gracias
Quisiera ayudarte con la documentación un poco para mejorar el proceso.
No estoy seguro cual es tu idioma nativo.
Lo que está escrito en Ingles no esta bien escrito y lo que está escrito en
Español tampoco es lo correcto.

richonguzman, 2024-01-22:
hola, por donde te escribo para esto?

yo hablo Español y el ingles es secundario