richonguzman/LoRa_APRS_Tracker #46: Iphone bluetooth issue

Bitbanger1900, 2023-12-09:
Hi,

I see on 2023.12.07 Added TTGO Lilygo LoRa32 v2.1 board as Bluetooth TNC(Android/Apple) and as a Tracker (with external GPS module).

I tried multiple Iphones with this TTGO  but i don't see the Bluetooth Lora device. On Android Phones it is working without problem.  Do i have to set a different Bluetooth type in the config?

Regards,
Johannes.

richonguzman, 2023-12-09:
Hi 

Apple works different:

You can't add a BLE unit on the BT part of preferences/configuration of the iPhone.

It only works when adding inside an app.

So for now it's working inside APRS.fi app when adding as a TNC

Bitbanger1900, 2023-12-09:
Hi,

Thanks  for your quick response! That is actually exactly what i did. Opened aprs.fi, go to TNC & DPS Modem, selelect a
TNC or software modem and than it keeps on searching but it finds nothing.

Regards,
Johannes

richonguzman, 2023-12-09:
There are a few things to test:

Reset or close APRS.fi app
Turn off and on BT on iPhone
Turn off and on tracker

Bitbanger1900, 2023-12-10:
Hi,

Tried all those but not working.

When i set bluetoothType to 0  in the tracker_config.json and call  BLE_Utils::setup(); in the void setup(); 
i do see the BLE TNC discovered?

richonguzman, 2023-12-11:
you should see the TNC to add it (but first, do you have the TNC and messages capability in aprs.fi )??

Bitbanger1900, 2023-12-11:
Hi,

Yes. i have the TNC and message capability in aprs.fi.

I added in the LoRa_APRS_Tracker.cpp file above the line WiFi.mode(WIFI_OFF); 
the line BLE_Utils::setup(); so that the BLE_Utils routine is always executed.

After that all is almost working fine except one thing.  When i open a console i see
that the message what is passed to the gateway over Lora is cutting of the SSID.
So i see my callsign and not callsign-14 passed to the iGate on the Tracker on the serial console.

Regards,
Johan

richonguzman, 2023-12-11:
1) maybe is an issue of filesystem upload as "BluetoothType":0 would force the BLE to work without any issues

2) explain me more about the callsign problem please:
Where do you have the missing ssid? On the serial output? On the trackers? On the phone?

Bitbanger1900, 2023-12-11:
1: i'm going to test.

2: within aprs.fi i configured my Station as OE2XZY-14. 

I connect the tracker to a serial output and have the following output. As you can see on the last line
it says: OE2XYZ> and i believe it should be OE2XYZ-14>

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[0;32m[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station[0m
[0;32m[INFO][Main] Version: 2023.12.07[0m
[0;32m[INFO][Main] Number of APRS Messages : 0[0m
[0;32m[INFO][Main] GPS disabled[0m
[0;32m[INFO][LoRa] Set SPI pins![0m
[0;32m[INFO][LoRa] Set LoRa pins![0m
[0;32m[INFO][LoRa] frequency: 433775000[0m
[0;32m[INFO][LoRa] LoRa init done![0m
[0;32m[INFO][BME]  BME280 Module not active in 'tracker_conf.json'[0m
[0;32m[INFO][BLE] Waiting for BLE central to connect...[0m
[0;32m[INFO][Main] WiFi controller stopped[0m
[0;32m[INFO]��u�Waiting for BLE central to connect...[0m
[0;32m[INFO][Main] No Keyboard Connected to I2C[0m
[0;32m[INFO][Main] CPU frequency set to 80MHz[0m
[0;32m[INFO][Main] Smart Beacon is: On[0m
[0;32m[INFO][Main] Setup Done![0m
[0;32m[INFO][BLE] BLE Client Connected[0m
OE2XYZAPFII0WIDE1-1WIDE2-1[0;32m[INFO][BLE Tx] OE2XYZ>APFII0,WIDE1-1,WIDE2-1:@184223h4953.33N/01807.58E[032/000aprs.fi for iOS!wrs![0m

richonguzman, 2023-12-11:
What is the SSID in the aprs.fi app?

Bitbanger1900, 2023-12-11:
OE2XYZ-14

richonguzman, 2023-12-11:
Could not find any station with this callsign on aprs.fi to see raw packets

Bitbanger1900, 2023-12-11:
correct. i changed it of course as i will not publish my callsign/GPS  in public on github

richonguzman, 2023-12-11:
Write me a telegram message at my github user to know your callsign and look at raw packets to help you further

richonguzman, 2023-12-13:
please download the latest "main" firmware and update as it should be corrected now

Bitbanger1900, 2023-12-13:
Working like a Charm! Thanks for your support!

Donation is on it's way and keep up the good work

closed 2023-12-13 by richonguzman