richonguzman/LoRa_APRS_Tracker #48: frequency in the profile

sp9svh, 2023-12-13:
Would be possible to add declarations of different frequencies and CR/SF parameters in each profile? In our country, we use these parameters other than generally agreed. This would allow us to quickly change them when we are near the border or when traveling to another country. It would be very useful for us and maybe also in other countries, for example in the UK.

sp9svh, 2024-02-29:
It's already working ! Thanks a lot! 73!

richonguzman, 2024-02-29:
great! tell me if this is working good for you ;)

sp9svh, 2024-03-01:
Yes, thank you, everything works great. A very useful feature. We are very grateful.
I only suggest adding information on the start screen (EU/PL/UK icon) which system is currently active - there will be no need to check in the menu. It probably won't fit on the main screen. BTW... I made a small modification to the previous SW by adding a few lines in tracker.config.json and configuration.cpp and connected a microswitch to GPIO2. When checking the pin status in void.setup(), I selected the required standard. A simple and effective solution, but now it is no longer needed 😊. In any case, the icon that I also added to the start screen was very useful to me at that time.
73! Krzysztof SP9SVH

richonguzman, 2024-03-01:
tell me about the icon.... maybe we could add some kind of startup info screen before running normally

like a status at boot up ?

sp9svh, 2024-03-01:
Maybe "icons: it's not good statment.. but I mean about an additional line between (tracker) and Richonguzman / CA2RXU in format for example:

       (tracker)
Standard: [EU] [UK] [PL]
Richonguzman / CA2RXU

In my opinion  it shuld be enough.

sp9svh, 2024-03-01:
it is not about choosing a standard, but only about presenting  information what standard is currently set.

sp9svh, 2024-03-02:
I have an idea :)
To show the current standard (EU-PL-UK) we can use the place where APRS symbol is located on the main screen.
In case when bluetooth is disconected :
The current APRS symbol and information about the current standard EU/PL/UK will be shown alternately.
In case when bluetooth is conected :
Bluetooth symbol and standard symbol will be shown altrernately - here, showing the APRS symbol makes no sense because it is determined by the connected device - (in my case aprsdroid).
For that is needed:  
- to add to custom_characters.h three new icons [EU/PL/UK] - (that's already done)
- to modify display.cpp so that appropriate icons are selected based on the current standard and displayed according to the assumptions above. ( I'm working on that).
What do You think about such solution ?

sp9svh, 2024-03-04:
Thanks a lot. The frequency standard is already visible on the screen during start, but there is one problem... It's always "EU" after restart. 
The setings is not remembered :( .  The callsign setting does not change after restart, but the frequency setting returns to the default value.

richonguzman, 2024-03-04:
this is odd the Freq loads right and stays as the last used freq...  but you are right, the Bootup message does not update... will fix this.

sp9svh, 2024-03-05:
I have uploaded and tested. Now it works great. At this stage the software is already brilant !
Thanks a lot for your effort. 
BTW. I have added to my tracker train icon from issue #52 by gersey  - very nice lokomotive. I often travel by train :)

richonguzman, 2024-03-05:
Train and yagi added

please check them out ;)

sp9svh, 2024-03-05:
Already tested. It works as needed :) 
I am considering an additional tracker mod but I am not sure if it is worth the extra effort as it would only be useful in a limited area near the border where access to digi gates operating at different frequencies is likely. in this case, the tracker listens on the primary frequency and after sending the beacon, it listens for a short while on the second frequency and transmits the same beacon there. After that it returns to the basic frequency. This functionality can currently be set in the DL9SAU software. This functionality is not needed for majority of users, but maybe it could be added later, when other more important functionalities are already introduced. It's an idea only :)

richonguzman, 2024-03-05:
its a good idea, but has some problem, when changing to different freqs , it could lost some packets sended to him....

this is why the change freq  is useful , but we cant be changing all the time the freq and expect to receive all the messages and packets sended to us.

although its a good idea as a solution for having Dif freq in nearby countries

sp9svh, 2024-03-05:
Thank you for the fast reponse. Of course, I am aware of these flaws. The battery will also be discharged faster... If someone wants to use the tracker just to track their position, this may be acceptable, but if want to receive messages etc , this is obviously problematic.
I thought of it as an option to consciously use in a specific situation. But as I wrote, this is only a functionality useful for a very limited number of users, so I think it will only remain in the area of ​​​​ideas :)

richonguzman, 2024-03-05:
lets save this idea for the future, maybe we could add this as a special feature later, meanwhile I will close this ;)

closed 2024-03-05 by richonguzman

sp9svh, 2024-03-06:
Hi Ricardo,Thanks a lot for your solution for frequency issue, it's a 
big gift for hams in Poland and probably not only. Of couse' it's good 
decision to close this topic - all problems at this stage are sucessfuly 
solved. In the meantime I contacted Damian SQ2CPA and we agreed to 
cooperate in testing his solutions for the igate/digi software.

If You need any support  during the tests  or anything else related to 
your project write  to me please  ***@***.***

Greetings from Poland Krzysztof (Chris) SP9SVH

W dniu 2024-03-05 23:59, Ricardo Guzman (Richonguzman) napisał(a):

> lets save this idea for the future, maybe we could add this as a 
> special feature later, meanwhile I will close this ;)
> 
> [2].

Links:
------
[1] 
https://github.com/richonguzman/LoRa_APRS_Tracker/issues/48#issuecomment-1979778975
[2] 
https://github.com/
--=_06188813cf910706b3774cd1f31f3b97
Content-Type: multipart/related;
 boundary="=_2c9a2165fab720a30c112c550cb6fce5"

--=_2c9a2165fab720a30c112c550cb6fce5
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hi Ricardo,Thanks a lot for your solution for frequency issue, it's a bi=
g gift for hams in Poland and probably not only. Of couse' it's good decisi=
on to close this topic - all problems at this stage are sucessfuly solved. =
In the meantime I contacted Damian SQ2CPA <span class=3D"HwtZe"><span class=
=3D"jCAhz ChMk0b"><span class=3D"ryNqvb">and we agreed to cooperate in test=
ing his solutions for the igate/digi software.</span></span></span></p>
<p><span class=3D"HwtZe"><span class=3D"jCAhz ChMk0b"><span class=3D"ryNqvb=
">If You need any support  during the tests  or anything else rel=
ated to your project write  to me please  <a href=3D"mailto:sp9sv=
<p><span class=3D"HwtZe"><span class=3D"jCAhz ChMk0b"><span class=3D"ryNqvb=
">Greetings from Poland Krzysztof (Chris) SP9SVH<br /></span></span></span>=
</p>
<p><br /></p>
<div id=3D"signature"></div>
<p><br /></p>
<p id=3D"reply-intro">W dniu 2024-03-05 23:59, Ricardo Guzman (Richonguzman=
) napisa=C5=82(a):</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto">lets save this idea for the future, maybe we could add this=
 as a special feature later, meanwhile I will close this ;)</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
ichonguzman/LoRa_APRS_Tracker/issues/48#issuecomment-1979778975" target=3D"=
_blank" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=3D"ht=
35AVCNFSM6AAAAABATZDY7SVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSNZZG43TQO=
JXGU" target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />Y=
ou are receiving this because you authored the thread.<img src=3D"cid:17097=
=3D"color: transparent; font-size: 0; display: none; visibility: hidden; ov=
erflow: hidden; opacity: 0; width: 0px; height: 0px; max-width: 0; max-heig=
/issues/48/1979778975</span><span>@</span><span>github</span><span>.</span>=
<span>com&gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_2c9a2165fab720a30c112c550cb6fce5
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_2c9a2165fab720a30c112c550cb6fce5--

--=_06188813cf910706b3774cd1f31f3b97--