richonguzman/LoRa_APRS_Tracker #49: Improved T-Beam V1.1 (UBlox M8N, SX1262, AXP2101) supported?

dc1rdb, 2023-12-19:
I just bought the above mentioned T-Beam with the improved LoRa transceiver. 
Tried several of your predefined build environments, but I keep getting LoRa starting errors:
```
[INFO][AXP2101] init done!
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2023.12.13
[INFO][Main] Number of APRS Messages : 0
[INFO][LoRa] Set SPI pins!
[INFO][LoRa] Set LoRa pins!
[INFO][LoRa] frequency: 433775000
[ERROR][LoRa] Starting LoRa failed!
```
Any tips how this configuration can be made to work with your tracker would be greatly appreciated.

Thanks in advance and 73 de Robert DC1RDB

[image]

richonguzman, 2023-12-19:
Hi Robert,

Write me at telegram (if you can) at my user (same as github user) and we could start adding code to work with your LoRa unit

I don't have a tbeam with sx1262 nor a single sx1262 Lora module to test here. So it could take a few messages back and forth but at least we could try ;)

richonguzman, 2023-12-19:
board added ;)

closed 2023-12-19 by richonguzman

dc1rdb, 2023-12-20:
Thank you so much, Ricardo. Works like a charm :)