richonguzman/LoRa_APRS_Tracker #51: Battery value not working anymore

iw2ejh, 2023-12-26:
Today I upgraded FW from version 2023.10.01 to version 2023.12.24  on my TTGO-T-Beam 1.2.
I noticed the battery value are not reported correctly as before:

previous value: 
2023-12-26 13:53:07 CET: [IW2EJH-5](https://aprs.fi/?c=raw&limit=&call=IW2EJH-5)>APLRT1,WIDE1-1,qAR,[iw2ejh-11](https://aprs.fi/?c=raw&limit=&call=iw2ejh-11):!/7?UxQ$YS[F6Q Bat=4.20V (215mA)
actual value:
2023-12-26 14:13:01 CET: [IW2EJH-5](https://aprs.fi/?c=raw&limit=&call=IW2EJH-5)>APLRT1,WIDE1-1,qAR,[iw2ejh-11](https://aprs.fi/?c=raw&limit=&call=iw2ejh-11):!/7?V\Q$YV[DvQ Bat=0.29V ( 0%)

 The battery is fully charded as shown in the beacon sent 20 minutes before the upgrade.

Also, the display says battery not connected and the blue LED (indicating the battery in charge mode) is permanently active, with or without the power supplied from the USB connector.

iw2ejh, 2023-12-26:
My mistake, I have a T-Beam V1 not V1_2,   I rebuild with the right environment and it display the correct values

iw2ejh, 2023-12-26:
could be closed, wrong environment selected.

closed 2023-12-26 by iw2ejh

richonguzman, 2023-12-26:
good to know you solved this ;)