richonguzman/LoRa_APRS_Tracker #52: New symbol

gersey, 2023-12-26:
I have created three symbols that I use. Put them in the code if you like.

// "="
static const unsigned char trainSymbol[] PROGMEM = {
	0b00000000, 0b00110000,
	0b11111110, 0b00000000,
	0b10000100, 0b01111000,
	0b10000100, 0b00110000,
	0b10000100, 0b00110000,
	0b10000100, 0b00110000,
	0b11111111, 0b11111100,
	0b11111111, 0b11111110,
	0b11111111, 0b11111110,
	0b11111111, 0b11111100,
	0b11100111, 0b10011110,
	0b01001111, 0b11001111,
	0b00100111, 0b10010000,
	0b00011000, 0b01100000
};

// "y"
static const unsigned char yagiSymbol[] PROGMEM = {
	0b01010000, 0b00000000,
	0b11110001, 0b10000000, 
	0b01000111, 0b11100000, 
	0b01011111, 0b11111000, 
	0b01111111, 0b11111110,
	0b11111111, 0b11111111,
	0b01000000, 0b00000010,
	0b01011000, 0b00011010,
	0b01011000, 0b00011010,
	0b01000011, 0b11000010,
	0b01000010, 0b01000010,
	0b01000010, 0b11000010,
	0b01000010, 0b01000010,
	0b01111111, 0b11111110
};

// "-"
static const unsigned char homeSymbol[] PROGMEM = {
	0b00100000, 0b00000000,
	0b00100001, 0b10000000, 
	0b00100111, 0b11100000, 
	0b00111111, 0b11111000, 
	0b01111111, 0b11111110,
	0b11111111, 0b11111111,
	0b01000000, 0b00000010,
	0b01011000, 0b00011010,
	0b01011000, 0b00011010,
	0b01000011, 0b11000010,
	0b01000010, 0b01000010,
	0b01000010, 0b11000010,
	0b01000010, 0b01000010,
	0b01111111, 0b11111110
};

richonguzman, 2023-12-26:
hello!

nice add!

can you share how they all look on your oled screen? (as pictures?) 

I see you have different house/home (and if it looks good I will replace mine with yours)

gersey, 2023-12-26:
[image]
[image]

In the meantime I saw that a house already existed, so I withdrew it. But the yagi is different, because of the antenna.

gersey, 2023-12-26:
// "L"
static const unsigned char lighthouseSymbol[] PROGMEM = {
	0b11000000, 0b00000011,
	0b00110001, 0b10001100,
	0b00001011, 0b11010000,
	0b00000101, 0b10100000,
	0b00000101, 0b10100000,
	0b00001011, 0b11010000,
	0b00110010, 0b11001100,
	0b11000011, 0b11000011,
	0b00000010, 0b11000000,
	0b00000111, 0b11100000,
	0b00000101, 0b11100000,
	0b00000111, 0b11100000,
	0b11100101, 0b11100111,
	0b11111111, 0b11111111
};

[image]

richonguzman, 2024-03-05:
train and yogi house added...  is this lightHouse pretty? :P

richtigerprashant, 2024-04-10:
hi friends, 
is it also possible to use some from the table which dont have a symbol on the device. eg. I would like to use the van, with k \ which anyhow breaks the json as it seems.

richonguzman, 2024-07-14:
train and bus added, dog still very bad

closed 2024-07-14 by richonguzman