richonguzman/LoRa_APRS_Tracker #54: Add usb TNC support

drtechga, 2023-12-29:
Hello would it be possiable to add USB tnc support to use with APRX as a multimode digirepeater? I am having issues with the bluetooth giving errors and disconnecting from raspberry pi using the rfcomm method.

Thanks,
KK4JMW

richonguzman, 2023-12-29:
Hello

If you give me more info about how or what is your goal connecting it , sure thing I can do it.

First: what is APRX ?

Do you need just serial output or else?

drtechga, 2023-12-29:
Hello, I building a mobile IGate/Digi, APRX is and advanced Linux software that allows you to Digi between multiple radios in diffferent configurations.

I'm looking for Serial TNC out of the USB cable.

Unless the I-GATE repository could be update to the tt-go beam with GPS support.

richonguzman, 2023-12-29:
1) soon my igate firmware will run with T-beam and gps board (hope soon)

2) what type of serial output do you need from the boards for APRX? Just the packet in a string?

drtechga, 2023-12-29:
1. Great Can't wait
2. Yes

drtechga, 2023-12-29:
I see this listed in the I-gate: 2023.10.08 Added Serial Comunication with PinPoint APRS app. Can this be enabled without digi/igate support? That would work as well.

Thanks

richonguzman, 2024-01-05:
this can be added or turned on in the "igate_conf.json" file. Please test if this is enough for you and let me know

drtechga, 2024-05-15:
Was added in Igate version

closed 2024-05-15 by drtechga