richonguzman/LoRa_APRS_Tracker #56: Auto power off after external power disconnected

gorzynsk, 2024-01-03:
It would be great if tracker sends position on external power off and shut downs.

Scenario: install it in a car, autostart on turning the key on. Then after parking send position od turning key off and clean shutdown.

gorzynsk, 2024-01-04:
Yes, exactly.

richonguzman, 2024-01-04:
But what would be the difference than using it without 18650 battery and shutting it off when the car is off ?

gorzynsk, 2024-01-04:
The difference would be in last report with speed 0 km/h and exact parking
spot.

richonguzman, 2024-01-04:
aha! now I get it, you want the last spot to be where you leave the Car/bike

what about just pushing the middle button to force Tx beacon ? or do you want a menu to save it as Last position and turn off automatic?

Are we always talking about T-Beams? as other boards won't allow for "turning off" from software command

gorzynsk, 2024-01-08:
Well, if the tracker would send a position at a parking spot automatically
I would mount it fixed in the car and react only if a message arrives. I
cannot trust myself to push a button each time (-: I'm so absentminded that
I forget to take the tracker from my car and it drains battery in my garage.

A... and yes - t-beams are most common as trackers here. The ones without
GPS and battery are used as digi/i-gate.

richonguzman, 2024-01-08:
maybe this?

if the tracker have not moved for 15min , it would send a beacon and turn itself off ?

or turn itself off after 3 or 4 not moving positions (at 15 min between beacons)?

gorzynsk, 2024-01-10:
Yes, it makes sense.

richonguzman, 2024-01-10:
So , what you propose:

Add a mode to turn itself off if not moving for 15 min?

gorzynsk, 2024-01-12:
Still the best user experience would be to send a beacon after power is
removed from USB port and turn off. Of course there may be a way to turn
that feature on and off.

Best regards
Kuba

valentintintin, 2024-03-22:
Well, nice feature that I was coding on my side but with a few bugs ...

- When board powered via USB : normal operation
- When powered via Battery : 
- 1 : if moving, normal operation
- 2 : if stopped for 5 minutes (set in JSON), power off for an amount of time (set in JSON), at wake up, we TX and check if we are moving --> 1 otherwise --> 2
- If possible, add an interrupt when USB is supplied to wake up without waiting for the sleep time out

In this case, I can leave the tracker in place, I can check the position of my car without thinking about the battery, switch it on etc

Michael-AUG, 2024-05-27:
I've just seen that this is the feature I was asking for, so I'll +1 this request and delete mine. Thanks :)