richonguzman/LoRa_APRS_Tracker #58: Changing timezone

sp2csr, 2024-01-04:
Is it possible to change the time zone? The current time in Poland is GMT +1. For ex. Local Time: 20:15, GPS Time: 19:15, and the GPS Time is sending in beacon...

richonguzman, 2024-01-05:
hi, the Time Zone is what you receive from GPS, which is UTC time

do you need a special custom Time mod for using in another GMT time zone?

sp2csr, 2024-01-06:
No no, i need info how to display correct time (for my Loc) on Tracker Display.

richonguzman, 2024-01-06:
will try to find something, but the data from GPS have always been UTC time , this is why only this is showed.

as far as I know there is no GMT correction available with this library, but will see if I can do something about it

iw2ejh, 2024-01-06:
May be you refer to the daylight saving time? My tracker show 12:35 while the APRS.FI report the beacon as 13:35..  Now the time here is 13:35.  During summer, the times are aligned.  https://en.wikipedia.org/wiki/Daylight_saving_time_by_country

richonguzman, 2024-01-06:
yes! satellites only send UTC time, they dont care about your position or daylight time savings. This is a local thing and should be handled this way.

looking for a code mod to enable some corrections about it

Szetya, 2024-01-21:
Maybe the function setenv() for the system time. It can handle time zones and daylight saving time.

richonguzman, 2024-01-22:
i don't find it to clear to implement, besides looks not updated for long time.

should this be handled on a self build library o function?

Szetya, 2024-01-22:
Now I look, this function belongs to the NTP library. Unfortunately, since the tracker has no internet connection, it is not good. :(

closed 2024-01-22 by richonguzman