richonguzman/LoRa_APRS_Tracker #62: Android Bluetooth not connecting

cshoaf, 2024-01-23:
Bluetooth will not pair with Android devices.  Bluetoothtype is set to 1, but does not show up when scanning for device.  Tried it with three different android devices. Tried setting Bluetoothtype to 0 and it shows up, but obviously won't work with Android devices.  Any idea?  Using a Heltec v3.

richonguzman, 2024-01-25:
it seems Heltec v3 only has BLE and not BT (classic) so unless the Android has BLE it seems you are out of luck

aximm, 2024-02-18:
I'm haveing exactly the same issue. Heltec v3 doesn't show up when scanning for devices. My phone supports BLE.

richonguzman, 2024-02-18:
The android part is build with BT classic in mind

I will have to check what to change to use this in android over BLe

miodzicho, 2024-03-07:
Same with LilyGo T-Beam 32Core SUPREME. Not visible in android BT scan.

richonguzman, 2024-03-07:
is your Android BLE or BT classic?

miodzicho, 2024-03-07:
BT5.3 LE so I presume it is BLE

richonguzman, 2024-05-09:
Just found out APRSDROID only uses SPP and ESP32S3 boards cant do it.... I will still keep on looking

closed 2024-05-09 by richonguzman