richonguzman/LoRa_APRS_Tracker #65: support for Heltec Wireless Tracker

lachutm, 2024-02-15:
Any chance to add support for this board:
[image]
it's Wireless Tracker kit based on ESP32-S3FN8 with 0.96 inch 160 * 80 LCD display. it runs on SX1262 LoRa chip and UC6580 for GPS.

richonguzman, 2024-02-15:
Hi,

I would love to add this...

can you help me with needed data to start trying and testing?

like Pinouts and such?

lachutm, 2024-02-15:
[Wireless Tracker 1.1.pdf](https://github.com/richonguzman/LoRa_APRS_Tracker/files/14297551/Wireless.Tracker.1.1.pdf)
 Most of this is available on their web https://heltec.org/project/wireless-tracker/

lachutm, 2024-02-15:
Let me know if You need testing on hardware.

richonguzman, 2024-02-15:
ok, this should be called "HELTEC WIRELESS TRACKER" ?

I found the pinout for:
-LoRa (8,9,10,11,12,13,14)
-GPS / GNSS (33,34,35,36)  (we need to look at how to make it work!)
-BatteryPin 1

- TFT ( 21, 38,39,40,41,42) the TFT could wait for the last part..

do you have telegram to make this faster? (look at my github user at Telegram)

HA7CQ, 2024-02-18:
I have this panel work well with meshtastic. (GPS work inside the building :) )
When you support this I can help you  to start trying and testing!

lachutm, 2024-02-18:
Hello I have a two hours now for testing. Letter afternoon I'm leveling, and I will be back late evening.

Szetya, 2024-02-20:
First test. It needs a bit of tweaking. 😇
[image]
[image]

Szetya, 2024-02-22:
https://github.com/Szetya/LoRa_APRS_Tracker

HA7CQ, 2024-02-23:
I install it to the heltec wireless tracker 433Mhz panel. Running Wel! Thanks a lot Szetya! 
[image]

kratinoska, 2024-03-04:
> https://github.com/Szetya/LoRa_APRS_Tracker

page does not exists.

Szetya, 2024-03-05:
> > https://github.com/Szetya/LoRa_APRS_Tracker
> 
> page does not exists.
Perhaps not by chance.

richonguzman, 2024-04-12:
Heltec Wireless Tracker support added ;)

closed 2024-04-12 by richonguzman

HA7CQ, 2024-04-17:
Hi Ricardo!
I install the last main version to Heltec wirelles tracker.
Work well!
I experienced an error in the display, the symbol does not appear (truck) symbol = "k". Look at the attached photo. 
Thanks your effort!

 

Closed #65 <https://github.com/richonguzman/LoRa_APRS_Tracker/issues/65>  as completed.

richonguzman, 2024-04-17:
hi, the symbols are not displayed, but will be very soon as I have to learn a way to do it on TFT ;)

HA7CQ, 2024-04-17:
Ok, be good work!

Szetya, 2024-04-17:
😉
https://youtu.be/DllqYauCWqw?si=-zTdJIhhCmT1JYp9

richonguzman, 2024-04-18:
symbols added ;)