richonguzman/LoRa_APRS_Tracker #69: Soporte para otras placas (T-beam Supreme, T-Deck)

sysadminpower2019, 2024-02-24:
Hola como te va.
Veo que ahora hay mas opciones que nunca para LORA y quisiera saber si fuera posible tener soporte para estas nuevas placas tambien. 
Aqui algunos vinculos a los vendedores en aliexpress
https://www.aliexpress.com/item/1005005692235592.html?algo_exp_id=31c4ce0f-7b20-4845-8d3d-8e018a939e6c-11&utparam-url=scene%3Asearch%7Cquery_from%3A
https://www.aliexpress.com/item/1005005975847274.html?algo_exp_id=65cb9faf-0ef4-47aa-a842-709e00128f9c-12&utparam-url=scene%3Asearch%7Cquery_from%3A

richonguzman, 2024-02-24:
Hola

Estoy intentando aprender del código para poder usar esa también pero es bastante más complicada que el tracker

richonguzman, 2024-04-16:
tienes una T-deck para probar?

Szetya, 2024-04-23:
https://youtu.be/FUyCoIiv4gM?si=U9tsrr4c1efssJgE
The display design later... 😉

richonguzman, 2024-04-23:
> https://youtu.be/FUyCoIiv4gM?si=U9tsrr4c1efssJgE The display design later... 😉

tomorrow I'll share mine :P

closed 2024-05-09 by richonguzman

leonardinux, 2024-05-26:
how do you select the T-DECK device here?

[image]

richonguzman, 2024-05-26:
hi,

you just add the same boards from guide (TTGO .......)

and after that you change the environment  on VSCODE and it will update and download all the libraries for you