richonguzman/LoRa_APRS_Tracker #76: Low or no GPS lock.

Ciumaros, 2024-03-19:
Hello!!! After updating the fw to the latest revision I have noticed an poorly gps lock.
Before fw update, the t-beam got a gps lock in 1÷3 minutes maximum after power on and now it passes 20-30 minutes. I usually comute to work and the comute lasts about 20 minutes, therefore now it does not even starts transmitting and I have reach my destination.

richonguzman, 2024-03-19:
Hi,

can you tell me which version of T-beam do you have there on the board and which Environment did you use to upload it?

Ciumaros, 2024-03-19:
Yes, the board is t-beam t22_v1.1 and I have compiled and uploaded the fw with visual studio and platform io latest versions

richonguzman, 2024-03-20:
ok, but did you use the environment for the "ttgo_t_beam_v_1" ?

Ciumaros, 2024-03-20:
Yes!
Now I notice new simptoms behaving behaving weerd, even if it gets gps signal. It tx for first beacon then no tx except manual. Tonight I will try to recompile the fw and uploaded it and if continue I will revert to old version of fw... To identify device problem or fw problem.

richonguzman, 2024-03-20:
if you dont for more than 50 or 100mts (depending on the settings of the tracker) it will only send one beacon each 15 min.

Szetya, 2024-03-21:
> Yes! Now I notice new simptoms behaving behaving weerd, even if it gets gps signal. It tx for first beacon then no tx except manual. Tonight I will try to recompile the fw and uploaded it and if continue I will revert to old version of fw... To identify device problem or fw problem.

The configuration is outdated. It is no longer snart_beacon but smartBeacon that stores beacon information. Update the contents of the data directory and upload it.

closed 2024-04-23 by richonguzman