richonguzman/LoRa_APRS_Tracker #79: SX1262 output power too low

formtapez, 2024-03-23:
I measured the output power from my TTGO_T_Beam_V1_2_SX1262. The power level seems limited at only 10 dBm, not 22 dBm according to the datasheet.
After looking around in the code, i found that the current limit for the PA is not increased from its default value.
When setting it to its max. value using `radio.setCurrentLimit(140);`, the measured power increases from 10 to 17 dBm. (still not 22 dBm, but maybe my board is not that good)

Szetya, 2024-03-24:
Yes, this (too) is indeed missing from the code.
Try to raise the current limit higher (maximum value 240mA).
Do not raise the output power higher than 20dBm because +2 is added within the code. If you set a value higher than 20 in the configuration you will limit the radiolib back to 10dBm.
HA5SZI

tk5ep, 2024-03-24:
From my experience, NONE of the Lilygo modules do reach the advertised output power !
+17dBm is the max I got from a SX1278 and +20dBm from a SX1272

I've written a small test software for the Lilygo modules :
[https://www.egloff.eu/](url)

closed 2024-04-18 by richonguzman