richonguzman/LoRa_APRS_Tracker #81: feature suggestion: Compressed Position Report Data Format

iw2ejh, 2024-03-29:
Is there any plan to introduce compressed format in order to reduce the length of the payload, as in the future, increasing the number of trackers, the air time may get saturated quickly and reducing it for each tracker may be beneficial for the network.

Mane76, 2024-03-29:
Hello iw2ejh,

compressed position report data is included since the beginning and additionally MIC-E since 12/2023 in the tracker firmware (Position, Altitude, Speed, Beaing).

**Default:**
CALLSIGN>APLRT1,WIDE1-1:!/5ilCR,'&kJCQ
**MIC-E**
CALLSIGN>T8UU20,WIDE1-1:`(&EppA>/"7b}

BR
Manfred

iw2ejh, 2024-03-29:
Thanks for the info, I  made a mistake, I was asking for the iGate, I saw the beacon sent to APRS-IS is in clear txt,  I do not know if the beacon sent via RF is clear txt or compressed: 
here what I saw via aprs-is > 
2024-03-28 13:17:11 CET: [IW2EJH-13](https://aprs.fi/?c=raw&limit=&call=IW2EJH-13)>APLRG1,TCPIP*,qAC,T2LAUSITZ:=4549.48NL01004.29E_.../...g...t048r...p...P...h91b10020LoRa_APRS_iGate Solar powered + WX (Batt=3.22V

Mane76, 2024-03-29:
No problem - every comment is welcome.
From iGate the beacon to aprs-is as well as to RF is not compressed.

richonguzman, 2024-03-29:
> No problem - every comment is welcome. From iGate the beacon to aprs-is as well as to RF is not compressed.

for now ;)

richonguzman, 2024-04-12:
> Thanks for the info, I made a mistake, I was asking for the iGate, I saw the beacon sent to APRS-IS is in clear txt, I do not know if the beacon sent via RF is clear txt or compressed: here what I [...]

iGate has now compressed position ;)

closed 2024-04-12 by richonguzman