richonguzman/LoRa_APRS_Tracker #83: Date and File Error

JimlyTheReal, 2024-04-04:
Hi OM's,

i installed the last Version 2024.03.22 of the LoRa_APRS_Tracker on my LiLyGo T-Beam 1.2.

After some initial difficulties now it works, two problems are now my problem:
(Positions are correct, altitude and speed too and are transmitted correctly)
1. The date is not updated, the initial value always remains
2. After a few days of regular operation, the display shows Error in the middle of operation. The settings in the file system image are gone? A re-upload of the Filesystem Image fixes the error.

richonguzman, 2024-04-04:
hi, I will upload soon a GPS reset fix for this... If not available in two days , write me!!! as its almost ready

on the second topic, what error did you get? NOCALL on one of the saved callsign?

JimlyTheReal, 2024-04-05:
1. Thnx fr ur answer and the tip about the GPS fix. I did that quickly. Hurray, the date is now here. I thought there would be no data without GPS Fix, but it was just the date.

2. Yes, after the error is displayed, NOCALL is assigned.

closed 2024-04-05 by richonguzman