richonguzman/LoRa_APRS_Tracker #84: Crash on more then 4 configured beacons

aendes, 2024-04-16:
Hello,

i tried to setup the latest version(2024.04.12) in a new setup once with 4 and with 5 beacons.
If i configure 4 beacons everything works like expected but if i configure 5 beacons the ttgo-t-beam_v1_2 starts in a boot crash loop.
I tried to reorder and remove a different entry from the json and for me the only differences seems to be the amount of configured beacons, which cause the crash.

Error on console: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

What else could i help to clarify it this is en error/bug?

Andreas

richonguzman, 2024-04-16:
this is no error nor bug.... the internal memory of the board holds so much until it won't work....  so its limited to 4 beacons for now

do you need more?

aendes, 2024-04-17:
I tried to add a fifth beacon. My Idea was to have to different Tracker with same config file and just change the beacon settings after flashing.
So for my Problem it could also a solution to have more beacons configured but make it active in the config file and the flash. Then i would have the same file with different 4 beacons active.

Thanks for your great work

richonguzman, 2024-04-17:
try to make the configuration file bigger to test if it holds 5

aendes, 2024-04-18:
Could you explain to me howto make the configuration file bigger?

aendes, 2024-04-18:
Thanks for your help. With 3000 i can add 5 beacons

closed 2024-04-18 by aendes