richonguzman/LoRa_APRS_Tracker #86: "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled." after SPI init

skymaster64, 2024-04-20:
I'm trying to get a tracker to run with the board t-beam V1.2. Unfortunately so far without success.
I installed VSCode and Platformio on a fresh Win11 according to the instructions. Then imported the code according to the instructions, changed the call, selected the board "ttgo-t-beam-V1_2", let the build run and flashed the board.
Everything's nice up to here.
The board boots and shows the start screen with LoRa Freq [EU] and version 2024.04.18 on the OLED.
However, there is a bootloop.

Here is the log on the serial port
`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (752) SPIFFS: mount failed, -10025
[     8][E][SPIFFS.cpp:89]�egin(): Mounting SPIFFS failed! Error: -1
[INFO][AXP2101] init done!
[  1028][E][vfs_api.cpp:24] open(): File system is not mounted
[  1029][E][vfs_api.cpp:24] open(): File system is not mounted
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2024.04.18
E (5054) SPIFFS: mount failed, -10025
[INFO][Main] Number of APRS Messages : 0
[INFO][Main] Number of Winlink Mails : 0
[INFO][LoRa] Set SPI pins!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dbff4  PS      : 0x00060a30  A0      : 0x800d278c  A1      : 0x3ffd0260
A2      : 0x3ffc4dec  A3      : 0x3ffc6450  A4      : 0x3ffc4b90  A5      : 0x3f4022b8  
A6      : 0x3ffc6b8c  A7      : 0x00000000  A8      : 0x800dbfe9  A9      : 0x3ffd0240  
A10     : 0x3ffd02ac  A11     : 0x3f4022b8  A12     : 0x00000017  A13     : 0x0000001a
A14     : 0x00000012  A15     : 0x0000e100  SAR     : 0x00000005  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4011563d  LEND    : 0x4011564c  LCOUNT  : 0x00000000  

Backtrace: 0x400dbff1:0x3ffd0260 0x400d2789:0x3ffd02e0 0x40116506:0x3ffd03a0

  #0  0x400dbff1:0x3ffd0260 in LoRa_Utils::setup() at src/lora_utils.cpp:129 (discriminator 4)
  #1  0x400d2789:0x3ffd02e0 in setup() at src/LoRa_APRS_Tracker.cpp:180
  #2  0x40116506:0x3ffd03a0 in loopTask(void*) at C:/Users/xxxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 00db70a7b17046f5

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[INFO][AXP2101] init done!
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2024.04.18
[INFO][Main] Number of APRS Messages : 0
[INFO][Main] Number of Winlink Mails : 0
[INFO][LoRa] Set SPI pins!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dbff4  PS      : 0x00060a30  A0      : 0x800d278c  A1      : 0x3ffd1370
A2      : 0x3ffc4dec  A3      : 0x3ffc6450  A4      : 0x3ffc4b90  A5      : 0x3f4022b8  
A6      : 0x3ffc6b8c  A7      : 0x00000000  A8      : 0x800dbfe9  A9      : 0x3ffd1350
A10     : 0x3ffd13bc  A11     : 0x3f4022b8  A12     : 0x00000017  A13     : 0x0000001a  
A14     : 0x00000012  A15     : 0x0000e100  SAR     : 0x00000005  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4011563d  LEND    : 0x4011564c  LCOUNT  : 0x00000000  

Backtrace: 0x400dbff1:0x3ffd1370 0x400d2789:0x3ffd13f0 0x40116506:0x3ffd14b0

  #0  0x400dbff1:0x3ffd1370 in LoRa_Utils::setup() at src/lora_utils.cpp:129 (discriminator 4)
  #1  0x400d2789:0x3ffd13f0 in setup() at src/LoRa_APRS_Tracker.cpp:180
  #2  0x40116506:0x3ffd14b0 in loopTask(void*) at C:/Users/xxxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 00db70a7b17046f5

Rebooting...
`
**What am I doing wrong?**

I tried to find the location of the crash in the code.
This seems to be here:
"long freq = currentLoRaType->frequency;" in lora_utils.cpp
If I put debug output in front of this line I can see it.
If I set the debug output afterwards I no longer see it.
Unfortunately I don't know much about C++. My knowledge comes from Z80 Assembler and TurboPascal3 :)

What is noticeable is that SPIFFS cannot be mounted when it is started for the first time. 
I suspect that the line in question is trying to read the frequency from the SPIFFS, which fails.
An attempt by me to set the frequency statically lacks my knowledge.
But I probably would have just come to the next post with a similar call.

Kind regards, Ron

richonguzman, 2024-04-20:
Hi Ron, did you upload also the Filesystem in VSCODE??? usually this boot loop comes when its not uploaded and the board crashes as it has no info to start with

skymaster64, 2024-04-20:
Plaese excuse my mistake.
I found the necessary description in another (old) ESP32-APRS project.
I didn't see the important step in your instructions.
Maybe it is possible to point out this step at startup if the file system is not available?

73 de Ron

closed 2024-04-20 by skymaster64