richonguzman/LoRa_APRS_Tracker #87: Add instructions for programming via terminal / command line

NeoFromMatrix, 2024-04-21:
It would be nice if someone could add instructions on how to compile, program and upload the config via linux command line /terminal.

I'm not sure if there is a generic (cross platform) way via platformio.

1. Compile firmware (e.g. $pio run -e ttgo-t-beam-v1_2)
2. Upload firmware (via pio or esptool?)
3. Upload filesystem
    -  generate SPIFFS (maybe via Espressivs spiffsgen.py?)
    - upload SPIFFS.bin

Anyone got any hints?

richonguzman, 2024-04-21:
Hi ,  I would love to !

but I dont know much "command line" commands for vscode/esptool or the ways to do it.

but I would love to learn and apply this

richonguzman, 2024-10-01:
im adding very soon webflasher with web configuration page to avoid VSCODE and terminal flashing ;)

closed 2024-10-01 by richonguzman