richonguzman/LoRa_APRS_Tracker #91: Crashes on a Lilygo TBeam

OK9UWU, 2024-04-27:
Hi, i wanted to try the firmware on my Lilygo Tbeam (not exactly sure which version) with SX1262 radio and AXP192 charge controller.
I had to modify the firmware to allow being use with this board, however when i uploaded it, it was periodically crashing at "Set SPI Pins", bunch of error logs 
[image]
and a discriminator 4 error (at set freq?) 
[image]
after which the esp32 just reseted itself.

At first i was suspicious of my mods to the firmware, but after uploading a version for a different board (eg. ttgo-t-beam-v1_2_SX1262 which is using a different charge controller) i was still getting the same error, no matter what build env.

Any ideas,  perhaps i forgot to update something, but that should be possible with platformio handling everything.
Tnx

richonguzman, 2024-04-28:
Hi

Can you post a picture of your board?

I have not seen a sx1262 module on a axp192 v1.1 board

Mylab0, 2024-04-29:
Hello,
I had exacly the same problem with a standard Tbeam v1.1!!
This is was happened:
3 days ago I compiled the sources and installed it on a 1.1 board. All good.
The day after I tried the igate.
After that I tried to reinstall the tracker compiling the same source one more time (I pressed the V and then ->).
I noticed that Platformio updated some files.
As result this time the sources were compliled and uploaded like before but the board hangs on boot like described by ok9uwu.
I tried many times, also using another V1.1 board.
No way, both hang on boot.
In the same time both operate smooth with the igate fw (or other firmware like meshtastic).
In my modest opinion (I'm not a programmer) there's something related with an update of the platformio libraries that are used in the firmware...
Many thanks, 73!

richonguzman, 2024-04-29:
but are your boards V.1.1 or V1.2? 

please post picture of the boards (also that it clearly says SX1262) 

> Hello, I had exacly the same problem with a standard Tbeam v1.1!! This is was happened: 3 days ago I compiled the sources and installed it on a 1.1 board. All good. The day after I tried the igate. After that I tried to reinstall the tracker compiling the same source one more time (I pressed the V and then ->). I noticed that Platformio updated some files. As result this time the sources were compliled and uploaded like before but the board hangs on boot like described by ok9uwu. I tried many times, also using another V1.1 board. No way, both hang on boot. In the same time both operate smooth with the igate fw (or other firmware like meshtastic). In my modest opinion (I'm not a programmer) there's something related with a recently update of the platformio libraries that are used in the firmware... Many thanks, 73!

but are your boards V.1.1 or V1.2? 

please post picture of the boards (also that it clearly says SX1262)

Mylab0, 2024-04-29:
My boards (both) are 1.1. I haven't the boards now with me but I have the attached photo. As you can see they are provided with HDP14A that should be SX1278 based module in my understanding.
After the last programming it's rebooting continuously. To be honest I don't remember if it hang exactly in the same point of OK9UWU but I get "Guru meditation error" and it reboot after 2-3 sec.
As said, the first time (one day before) it was fine...
[image]

richonguzman, 2024-04-29:
> My boards (both) are 1.1. I haven't the boards now with me but I have the attached photo. As you can see they are provided with HDP14A that should be SX1278 based module in my understanding.
>
> After [...]

Did you upload the firmware and also the filesystem? When it reboots it is usually this.

Also when flashing in vscode you can see the serial output that helps a lot to debug

Mylab0, 2024-04-30:
I think I upload "everything" (bad word, I agree) since I compiled your sources with "v" buttom and then I pressed "->".  
I saw the continuous reboot because I connected the Putty terminal to the board (I could use the platformio terminal). 
Moreover since it was continuosly rebooting the display was dark.  In the previous compilation everything were good (no reboot and display ok).
In this moment I dont have the boards with me and I cannot indicate the exactly point were the boot hangs. I'll update you in two days as I can reconnect the boards to the terminal.
Thanks 73, IU5JPC

OK9UWU, 2024-04-30:
![IMG_20240501_005924](http
[image]
s://github.com/richonguzman/LoRa_APRS_Tracker/assets/61870387/6181ff84-68eb-4b89-8ee7-427511baae5e)

I believe mine is v1.1 aswell (and yes the board went to the hell and back with mods and whatnot).
I know that it's SX1262 for sure because i was using this exact board for some experiments with Jan Gromes' RadioLib.

OK9UWU, 2024-04-30:
Well, github managed to mess the second one up, let me upload it again.
[image]

richonguzman, 2024-04-30:
> Well, github managed to mess the second one up, let me upload it again. ![IMG_20240501_005924](https://private-user-images.githubusercontent.com/61870387/327009121-c8e249e6-5c7c-4ce5-a5e5-ede2b9ef9615.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1MTk0MjIsIm5iZiI6MTcxNDUxOTEyMiwicGF0aCI6Ii82MTg3MDM4Ny8zMjcwMDkxMjEtYzhlMjQ5ZTYtNWM3Yy00Y2U1LWE1ZTUtZWRlMmI5ZWY5NjE1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDIzMTg0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTliNzAwZWEzMDUwNDc0MjEzMGM4YjQ3NDUzYWZiODRlYmQ2ZTEwNmI3MzM3ZTk3NzIzNTdjZTQ2ODRjNTQ0NzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IjKsqPd96fuFB6POgX-72i3aE97-VJ3vZCtNTQ5ktZ8)

was is with the soldering of the lora module? was it changed?

i dont know if I have seen an V1.1 board with SX1262 yet...  

but before all of this, I need confirmation that when installing it all you uploaded the two steps  (firmware and then filesystem)

Mylab0, 2024-05-01:
Hello again!
Now I understand (!!), in my case the second time I programmed the boards, as you suggested, I missed to upload the filesystem (!) as I pressed only 'v' and '->'.
I have to check with the boards but 99.9% the problem is this one! Sorry for the mistake.
I'll follow up as I can check this on the board.
In the meantime apologize for the mistake and thaks for the support!
IU5JPC

OK9UWU, 2024-05-02:
Okay, yeah this works.
[image]
Could've occured to me earlier but oh well.

Uploaded the "ttgo-t-beam-v1_2_SX1262" env as thats closest to what i have onboard, just different battery managment chip. I can hack around it but i don't think i'll be able to add it to all the state machines i saw in the source code (i will try though).

richonguzman, 2024-05-02:
on platformio.ini change the details of this board from HAS_AXP2101 to HAS_AXP192 and maybe this is all what's needed

OK9UWU, 2024-05-02:
I believe there is more than that from when i tried it 5 days ago, because some ifdefs take the board env and some take in the the chip declaration, seamlessly trying to use both chips at once.

Anyway, i'll try it in a bit and report back.

Mylab0, 2024-05-03:
Hello,
all good from my side.
This time I uploaded both firmware and filesystem on the two board I have and both operates smooth! ..the reboot cause was the missed filesystem...
Many thanks for your support,
see you once on HF (or aprs!) 
IU5JPC

closed 2024-05-09 by richonguzman