richonguzman/LoRa_APRS_Tracker #92: LILYGO SX1262 T-Beam AXP2101 v. 1.2 problem

SP9MIB, 2024-05-01:
Hello,

After build, upload and upload filesystem image, I've got this on my terminal:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[INFO][AXP2101] init done!

The message loops and appears over and over again.

What is the problem? My device is: LILYGO SX1262 T-Beam AXP2101 v. 1.2 (without screen).

Best regards, Piotr

richonguzman, 2024-05-01:
Hi

Did you upload with the correct environment selection for this board?

SP9MIB, 2024-05-02:
Hi :) 

I choose PlatformIO and "ttgo-t-beam-v1_2_SX1262, upload and monitor.

richonguzman, 2024-05-02:
this both for firmware and filesystem ?

f4exb, 2024-05-08:
I had the issue until flashing the filesystem. It worked fine after this.

closed 2024-05-09 by richonguzman