richonguzman/LoRa_APRS_Tracker #93: Feature: Support RFM96 LoRa chip

petrkr, 2024-05-06:
I have custom board which using RFM96 modules. Since RFM96 is quite similar to SX127x there should not be lot of changes. I just making this as placeholder for future work on it.

I wanted show it today, but it does not work out-of-box, so before I will forgot I will make this as placeholder to remember.

now I am in boot loop

```
[INFO][LoRa] Set SPI pins!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400db920  PS      : 0x00060a30  A0      : 0x800d2aec  A1      : 0x3ffd1740  
A2      : 0x3ffc4db4  A3      : 0x3ffc63b8  A4      : 0x3ffc4b58  A5      : 0x3f4022db  
A6      : 0x3ffc6ba4  A7      : 0x3f426863  A8      : 0x00000000  A9      : 0x3ffd1720  
A10     : 0x3ffc63b8  A11     : 0x0000001a  A12     : 0xffffffff  A13     : 0x0000001b  
A14     : 0x0000001a  A15     : 0x0000e100  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4010dd75  LEND    : 0x4010dd84  LCOUNT  : 0x00000000  

Backtrace: 0x400db91d:0x3ffd1740 0x400d2ae9:0x3ffd17c0 0x4010eb9e:0x3ffd1880
```

richonguzman, 2024-05-06:
So this is a "new" (old) board to add to tracker possibilities , right?

It seems that I runs over Radiolib and the pinouts are similar to sx1278 boards

do you have success using it on any custom code (not meshtastic?) to validate this fast?

petrkr, 2024-05-06:
Something like that. I have development boards before ttgo or others existed and I make break out board for rfm9x modules. And since I wanted to show aprs to one friend, who asked for it. I decide to try it on mine board as I have them and do not wanted wait 1 month for aliexpress one.

I used those boards back to 2018/2019 on TTN lorawan and later on HNT/Helium

Here I was using LMIC library with it
https://github.com/petrkr/helium-lora-sht3x

richonguzman, 2024-05-06:
> Something like that. I have development boards before ttgo or others existed and I make break out board for rfm9x modules. And since I wanted to show aprs to one friend, who asked for it. I decide [...]

which pins definitions change did you change on yours to make it work?

or what did you try (and change) to know where to start(over)

petrkr, 2024-05-06:
Here I did fork and created new board definition

https://github.com/petrkr/LoRa_APRS_Tracker/commit/0752172b6dd1b3417398954dc4d3442c1a9f1b2a

richonguzman, 2024-05-06:
> Here I did fork and created new board definition
> 
> [petrkr@0752172](https://github.com/petrkr/LoRa_APRS_Tracker/commit/0752172b6dd1b3417398954dc4d3442c1a9f1b2a)

this fork got the reboot loop?

but did you upload filesystem? as Lora pins definitions dont produce reboot loops :P

petrkr, 2024-05-06:
Filesystem? Nope :)

Time to read carefully build instructions. I had like half hour, so I did platform build and upload.

Now I am on way to HAM meetup. I will maybe have some time there too.

De OK1PKR

richonguzman, 2024-05-06:
Check this :

[tutorial-wiki](https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/04.-Upload-Firmware-and-Filesystem-%23-Subir-Firmware-y-sistema-de-archivos)

petrkr, 2024-05-06:
Well. Bootloop fixed, really flashing filesystem helps.

But for now really does not initialized that LoRa Radio

```
[INFO][LoRa] Set SPI pins!
[ERROR][LoRa] Starting LoRa failed!
```
[image]

richonguzman, 2024-05-06:
> Well. Bootloop fixed, really flashing filesystem helps.
> 
> But for now really does not initialized that LoRa Radio
> 
> ```
> [INFO][LoRa] Set SPI pins!
> [ERROR][LoRa] Starting LoRa failed!
> ```
> 
> ![IMG_20240506_173802732_HDR](https://private-user-images.githubusercontent.com/6428351/328234994-c071d386-c339-4626-8f4a-3dc12a31d20a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwMTAzNzgsIm5iZiI6MTcxNTAxMDA3OCwicGF0aCI6Ii82NDI4MzUxLzMyODIzNDk5NC1jMDcxZDM4Ni1jMzM5LTQ2MjYtOGY0YS0zZGMxMmEzMWQyMGEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDZUMTU0MTE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTA2MjA2NzU5Mjc4N2MwODA5Y2Y2MzU1NzlmZTZjNTUxZjFkNmYyMzg5NDcxNTA5NTQyYjYzM2Q3ZTYzMDM2MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.XAVUF9p2OS4Vy8AyRxohV7_qZRHPW5DC40abKxTPoMA)

time to revalidate pins definitions on pins_config.h

petrkr, 2024-05-06:
I have to test some test code for that module to check if module itself works. Pinout is fine, checked again.

I have to check how to add some more debug output/messages. So I can try RAW SPI messages.

Anyway thank you

richonguzman, 2024-05-06:
> I have to test some test code for that module to check if module itself works. Pinout is fine, checked again.
> 
> I have to check how to add some more debug output/messages. So I can try RAW SPI messages.
> 
> Anyway thank you

let me know how the testing goes, to add this also as a valid board later ;)

closed 2024-10-01 by richonguzman