richonguzman/LoRa_APRS_Tracker #94: Error code "failed, code -705"

Vitaliy86, 2024-05-07:
Error if send from bluetooth: failed, code -705
Recive not work (len on pcb on "2 pin" is off)

My config
`#ifdef ESP32_DIY_1W_LoRa_GPS   // My Ebyte E22 400M30S OR RA-01S / SX1268

#define OLED_SDA            21
#define OLED_SCL            22
#define OLED_RST            -1

#define GPS_RX              17
#define GPS_TX              16
#define GPS_PPS             26
#define BUTTON_PIN          -1//15
#define BATTERY_PIN         36  //ADC0  

#define RADIO_SCLK_PIN  18     // GPIO5    - SX12xx SCK ----> SCLK
#define RADIO_MISO_PIN  19     // GPIO19   - SX12xx MISO
#define RADIO_MOSI_PIN  23     // GPIO27   - SX12xx MOSI
#define RADIO_CS_PIN    5      // GPIO18   - SX12xx CS ---> NSS
//#define RADIO_DIO0_PIN  -1
#define RADIO_RST_PIN   14     // GPIO14   - SX12xx RST ---> NRST
#define RADIO_DIO1_PIN  33
#define RADIO_BUSY_PIN  39     // GPIO26   - SX12xx IRQ ----> DIO0  
#define RADIO_RXEN      2
#define RADIO_TXEN      4

#endif`
ps work in igate

Vitaliy86, 2024-05-07:
I change in tracker_config.json all pins in "-1" 
it works now

richonguzman, 2024-05-07:
Hi Vitaliy

I need more context and explanations what are you using, which board? or full diy? tracker firmware or igate firmware

all data needed please

Szetya, 2024-05-07:
> Megváltoztattam a tracker_config.json összes pinjét a "-1"-ben, ez most működik

In Config, the flashlight PIN conflicts with the RST PIN configuration of the lora radio.
"-705 timed out while waiting for complete SPI command."
-You cannot set the PIN to -1 because:
https://github.com/richonguzman/LoRa_APRS_Tracker/blob/e0373dbc156a08747e93e5fe6da0a82c81a1dcee/src/LoRa_APRS_Tracker.cpp#L162
this line should normally be like this:
if (Config.notification.ledFlashlight && Config.notification.ledFlashlightPin >= 0 ) pinMode(Config.notification.ledFlashlightPin, OUTPUT);
And other similar lines. :)

Szetya, 2024-05-07:
And RADIO_RXEN also conflicts with the ledmessagePin.

Vitaliy86, 2024-05-07:
**richonguzma** this project https://oshwlab.com/GabeH/esp32-lora-aprs

 tracker firmware 

**Szetya** yes, but work....

Szetya, 2024-05-07:
@Vitaliy86  😉
[image]

Szetya, 2024-05-07:
@Vitaliy86 
[image]

richonguzman, 2024-05-07:
> **richonguzma** this project https://oshwlab.com/GabeH/esp32-lora-aprs
> 
> tracker firmware
> 
> **Szetya** yes, but work....

so did you validate all pins from pins_config.h and also the notifications pins from tracker_conf.json to avoid being the same?

Szetya, 2024-05-07:
@richonguzman Almost all pins are used so you need - 1 option in the tracker_config.json file -1 for the PINs. But -1 can only be without error if the if() structure is as I wrote above. 
@Vitaliy86 Be careful with the pins_config and tracker_config.json files. I know several people who have ruined their radios because of this.
I had a pull request in February about this but withdrew it for lack of progress.

richonguzman, 2024-05-07:
> @richonguzman Almost all pins are used so you need - 1 option in the tracker_config.json file -1 for the PINs. But -1 can only be without error if the if() structure is as I wrote above. @Vitaliy86 [...]

why don't you add all the info needed for @Vitaliy86 , not just a basic guide?

PS: 

I added this validations for -1 values on tracker Fw this morning

Szetya, 2024-05-08:
> PS:
> 
> I added this validations for -1 values on tracker Fw this morning
@richonguzman  Yes, I see. The only problem is that it won't boot when -1.
In short: On the BV5DJ board, all the ESP PINs are busy = performs some function. So you need the possibility to replace the PIN for buzzer, ledtx, ledmessage, ledflaslight with a number (-1 is disabled by default) for which nothing happens.
So if a value less than 0 is specified, don't use that PIN (even if the corresponding value is enabled)
In this case the situation is now even worse than it was before.

richonguzman, 2024-05-08:
well, can you do a map of the pinouts needed and also the config values it needs for this?

> PS:
> I added this validations for -1 values on tracker Fw this morning
> @richonguzman  Yes, I see. The only problem is that it won't boot when -1.
> In short: On the BV5DJ board, all the ESP PINs are [...]

Szetya, 2024-05-08:
> Nos, meg tudod csinálni a térképet a szükséges pinoutokról és az ehhez szükséges konfigurációs értékekről?
> 
> > > PS: 
> > > Ma reggel hozzáadtam ezt az érvényesítést a -1 értékekhez az Fw trackerben
> > > @richonguzman  Igen, látom. Az egyetlen probléma az, hogy nem indul el, ha -1. 
> > > Röviden: A BV5DJ kártyán az összes ESP PIN-kód foglalt = valamilyen funkciót lát el. Tehát szükség van arra a lehetőségre, hogy a berregő, ledtx, ledmessage, ledflashlight PIN kódját lecserélje egy számra (a -1 alapértelmezés szerint le van tiltva), amelynél nem történik semmi. 
> > > Tehát ha 0-nál kisebb érték van megadva, ne használja azt a PIN-kódot (még akkor sem, ha a megfelelő érték engedélyezve van). 
> > > Ebben az esetben a helyzet most még rosszabb, mint korábban volt.

#ifdef ESP32_BV5DJ_1W_LoRa_GPS
#undef OLED_RST                 //Conflict with GPS TX pin
#define OLED_RST            -1  //Conflict with GPS TX pin
#define GPS_RX              17
#define GPS_TX              16
#define BUTTON_PIN          0   //ENT
#define RADIO_SCLK_PIN      18
#define RADIO_MISO_PIN      19
#define RADIO_MOSI_PIN      23
#define RADIO_CS_PIN        5
#define RADIO_RST_PIN       14  //NRST
#define RADIO_DIO1_PIN      33
#define RADIO_BUSY_PIN      39
#define RADIO_RXEN          2   //Same LED_BUILTIN
#define RADIO_TXEN          4
//EXTRA PINS HERE:
#define BUTTON_UP           34  //joystick UP
#define BUTTON_DOWN         35  //joystick DOWN
#define BUTTON_LEFT         27  //joystick LEFT
#define BUTTON_RIGHT        32  //joystick RIGHT
#define SD_CS               13  //MicroSD card SlaveSelect
#define GPS_PPS             26  //GPS PPS pin 
#define BATTERY_PIN         36  //ADC pin from voltage divider
#define RGB_LED_PIN         12  //WS2812 LED GPIO
#define LEDNUM              2   //WS2812 LEDs num's
#define KEEP_ALIVE          25  //Trigger -pad in PCB
#define DIO3_TCXO_REF       1.8 //DIO3 Reference Voltage
#endif

[image]

closed 2024-10-01 by richonguzman