richonguzman/LoRa_APRS_Tracker #95: posibilidad modificar hora UTC

sysadminpower2019, 2024-05-09:
Hola,

Queria preguntarte si no seria posibile de agregar una opcion para modificar la hora UTC que sale en el display.

Seria bueno tener la hora local ya sea via GPS o alguna opcion directamente en el modulo

Gracias

richonguzman, 2024-05-09:
Hola,

los gps solo envian hora UTC, no hay zonas horarias para la info que ellos envian.

sysadminpower2019, 2024-05-09:
Ah gracias por la info. Entonces quizas se podria agregar algun parametro en la programacion o en el menu para poder ajustar la hora

richonguzman, 2024-05-09:
esto es meramente cosmetico : lo dejare en las cosas por hacer, pero nada de lo pronto/urgente

closed 2024-05-09 by richonguzman

sysadminpower2019, 2024-05-12:
Si quieres mandar me tu discord para conversar un poco y ver si mejoramos la documentacion y el plan de trabajo

richonguzman, 2024-05-13:
puede ser por telegram ? me es infinitamente mas cómodo (porque no uso discord hahahaha) 

mi usuario es el mismo de aqui para que me encuentres ;)