richonguzman/LoRa_APRS_Tracker #96: tocall unexpected value

clwydian, 2024-05-09:
There is an excellent LoRa map link that shows both trackers and igates and nothing else.  However, it doesnt show my tracker.  This is because the filter requires a TOCALL of APL* and it seems my tracker has an TOCALL of US1PTV, US1PSW and possibly others but not APL...

Is this because I have enabled BME280 on my tracker, or am I doomed to obscurity for another reason?

LILYGO TTGO 1.2

(the map link BTW is https://lora.ham-radio-op.net/?center=51.5129,-0.9547&zoom=7 )

Carl
gw0tqm@yahoo.co.uk G0TQM-3 / GW0TQM-10

richonguzman, 2024-05-09:
Hi Carl,

it seems you enable MIC-E encoding which changes the tocall of the packet when encoding the gps data into it.

please confirm

clwydian, 2024-05-12:
Sorry for the delay.  Yes I did in the misguided attempt to shorten the packet length.   Thanks.

richonguzman, 2024-05-13:
great news all has been solved ;)

closed 2024-05-13 by richonguzman