richonguzman/LoRa_APRS_Tracker #97: new feature request

iw2ejh, 2024-05-11:
would it be possible to have the igate/digi function available on the tracker?  Heltec and TTGO

use case: I have the tracker in the car,  in an emergency scenario (but also in a mountain trekking), I then stop the car and deploy the pedestrian tracker. 

The people going around can be igated by the device in the car, if I switch from tracker to igate or digi so that the operation in the area around the car and tracked and passed to APRS-IS servers.

richonguzman, 2024-05-11:
Hi

Right now the tracker has an (emergency) mode for digirepeater if needed

Do you want an igate mode on the tracker? What about the internet connection and all of that?

iw2ejh, 2024-05-12:
Hi Ricardo,
I'm not sure what the Digirepeater/SOS mode is doing, if we can enable the digipeater without any SOS mode, just to use it as repeater then this is ok.(I didn't found a lot of info related to this mode, so not sure how it work)
For the Igate mode, on some emergency scenario (like SAR ) we have wifi available. On a single user scenario(mountain trekking), we can use a smartphone with wifi/tethering Router mode.

richonguzman, 2024-05-13:
ok , the emergency igate thing is good idea, but this will come later

for now, the Digi mode on the tracker is like you say , for emergency deployment of digi on a place:

you just go into Menu>6 Extras > Digirepeater (ON) and It will send beacons as a tracker but will repeat all packet that he hears also just like a digirepeater)

closed 2024-05-14 by richonguzman