richonguzman/LoRa_APRS_Tracker #100: Long TX on boot and manual TX on TTGO-T-BEAM

wisp-engineer, 2024-05-14:
Bug:
On first boot after unit gets valid GPS signal, first TX takes ~35 seconds to complete,
Not seen the unit do any further smart beaconing, doing a manual TX (central button short press) it does same where it takes 35s to TX

Expected behavior:
Normally TX takes approx 2-5s on average

[image]

Desktop
    OS: Win11 Pro
    PlatformIO Version: 1.89.1
    Firmware Version: 14-05-2024

Board name: LILYGO TTGO T-BEAM-v1_2

Board has had a complete erase and rebuild, still present

richonguzman, 2024-05-14:
this is very strange as I've been testing it here with V.1, V1.2, HELTEC WirelessTracker and other DIY boards and all handle it like a normal message

should we check in detail the filesystem configuration?

wisp-engineer, 2024-05-14:
Sure, can send over a copy of the conf.json for what I used for the tracker - no other variables anywhere else were changed, 
e.g. open VSS, created fresh download, let it update platform.io "as normal" , once settled added in the adjusted variables where needed (callsign, comment etc) most of the other settings left to default certainly for Freq and smart beaconing

richonguzman, 2024-05-14:
> Sure, can send over a copy of the conf.json for what I used for the tracker - no other variables anywhere else were changed, e.g. open VSS, created fresh download, let it update platform.io "as [...]

if you have telegram look for me (the same user as here on github) to see in detail your case

closed 2024-05-14 by wisp-engineer