richonguzman/LoRa_APRS_Tracker #102: Feature Request: Support for Web Configuration UI with WiFi AP

BD5HTY, 2024-05-17:
Support Web Configuration UI with WiFi AP.

73!!

richonguzman, 2024-05-17:
next week I will start coding for it!! ;)

BD5HTY, 2024-05-17:
Thank you for your contribution! I plan to use LILYGO®TTGO T-SIM7000G plus Lora module to form a 4G and Lora dual-mode tracker. I wonder if you are interested.
73！！

richonguzman, 2024-05-17:
did you see that I used a A7670 + ESP32 board  + SX1278 for iGate ???

I think this could be transformed into a tracker with GPS module!!

and on top of this, A7670 uses 4G/LTE and SIM7000 uses 3G or less , right?

BD5HTY, 2024-05-17:
> 你看到我为 iGate 使用了 A7670 + ESP32 板 + SX1278 吗？？？
> 
> 我想这可以改造成一个带 GPS 模块的追踪器！！
> 
> 除此之外，A7670 使用 4G/LTE，SIM7000 使用 3G 或更少，对吗？

You are right, It's outdated.I am very sorry that I mentioned the wrong model before. I will use the new 4G and GPS integrated module, AIR780EG.

BD5HTY, 2024-05-17:
> did you see that I used a A7670 + ESP32 board + SX1278 for iGate ???
> 
> I think this could be transformed into a tracker with GPS module!!
> 
> and on top of this, A7670 uses 4G/LTE and SIM7000 uses 3G or less , right?

You are right, It's outdated.I am very sorry that I mentioned the wrong model before. I will use the new 4G and GPS integrated module, AIR780EG.

BD5HTY, 2024-05-17:
I just checked the official information, T-A7670E does support GPS.

richonguzman, 2024-05-17:
yes it does!

the AT commands to do this is not that friendly but can work!

BD5HTY, 2024-05-18:
> did you see that I used a A7670 + ESP32 board + SX1278 for iGate ???
> 
> I think this could be transformed into a tracker with GPS module!!
> 
> and on top of this, A7670 uses 4G/LTE and SIM7000 uses 3G or less , right?

Is there any more information about A7670 + ESP32 board + SX1278 for iGate? Is the code already supported? How to connect SX1278?

richonguzman, 2024-05-18:
> Is there any more information about A7670 + ESP32 board + SX1278 for iGate? Is the code already supported? How to connect SX1278?

pins_config.h or boards_pinout.h has the pins used for this.

richonguzman, 2024-07-07:
> Is there any more information about A7670 + ESP32 board + SX1278 for iGate? Is the code already supported? How to connect SX1278?

how is it going changing the TOCALL to my firmwares?

pstankie, 2024-09-16:
I am wondering if there's any progress with Web UI + Wifi ? :-)

richonguzman, 2024-09-16:
> I am wondering if there's any progress with Web UI + Wifi ? :-)

very soon, was finishing RAK modules code