richonguzman/LoRa_APRS_Tracker #104: WX data? Comment? Looks like matrix :), TTGO Lilygo T-Beam S3 Supreme V3

Charly1000, 2024-05-19:
built on WX-data works perfect on built-in screen. 
But in example on the website aprs.fi it looks as you can see on the picture. 

[image]

richonguzman, 2024-05-19:
Hi

I would need to see your raw packets , what is your callsign and ssid used for this?

Charly1000, 2024-05-19:
yes, of course: DO3MMC-7

richonguzman, 2024-05-19:
> yes, of course: DO3MMC-7

can you change the MIC-E encoding to "" (yes, no MIC-E enconding at all)

I didn't test if sending Wx Packet with MIC-E encoding worked before

Charly1000, 2024-05-19:
MIC-E is full off. 
it doesn't work. Have a look on the raw data

richonguzman, 2024-05-19:
its working right now as I've checked

Charly1000, 2024-05-19:
You're right. 
But what's the code? .../...g...t080r.... ..?
and how it's possible to write own comments?

richonguzman, 2024-05-19:
you can add a Comment on /data/tracker_conf.json and upload filesystem again. (Check also that comments gets send once at X number of Tx packets. This number can be also changed in tracker_conf.json file as "sendCommentAfterXBeacons".

PS: 

I've update the code with this corrections on display ON and OFF, screen brightness and also to send the Wx Data at start and don't wait 10 min at first Wx data Tx

please update the code from github and test it all

Charly1000, 2024-05-19:
Perfect.
RESOLVED. 
it's working fine now

Thank you sooo much

richonguzman, 2024-05-19:
now Wx data won't be sended over Mic-E encoding even though all gps positions are in MiC-E

closed 2024-05-19 by richonguzman