richonguzman/LoRa_APRS_Tracker #105: Eco mode, TTGO Lilygo T-Beam S3 Supreme V3

Charly1000, 2024-05-19:
Eco-mode on/off has no impact. Screen is always on

richonguzman, 2024-05-19:
do hoy run your board with SH1106 or SSD1306 screen?

Charly1000, 2024-05-19:
It should be SH1106

[image]

richonguzman, 2024-05-19:
can you test modding the following?

/SRC/display.cpp  , line 132 on void display_toggle(bool toggle)...

I've added the "test code" lines to see if this works on SH1106 to turn screen on and off

[image]

Charly1000, 2024-05-19:
Perfect. RESOLVED.
it's working fine now.

closed 2024-05-19 by richonguzman